nightscout/AndroidAPS #2142: Dev-C cleaning database bug

robertrub, 2022-10-26:
@MilosKozak More info on the _clean db_ "reseting" db update to NS.

I have NS and Mongodb on my NAS in containers. All is working OK.

I pause NS update in AAPS. 
Set to zero the database on the NAS.
Clean db to 3 months in AAPS.
This sets the NS upload cue to over 55000
Unpause NS update and AAPS starts to send the 55+ k data.
Once all the data is uploaded, I go to my NS web page and try reports. I only have 1 days data. Theoretically, there should be 3 months data.

It is as if the data was not uploaded or was uploaded not as NS expects.

[image]

robertrub, 2022-10-26:
[image]

[image]

MilosKozak, 2022-12-04:
can you verify ns db via https://..../api/v1/treatments.json ?

robertrub, 2022-12-04:
I must be doing something wrong as I get this message
[image]

MilosKozak, 2022-12-04:
can you open NS and see data on this computer?

closed 2023-01-01 by robertrub