nightscout/AndroidAPS #2145: Dev-C Winter time / Prediction lines

robertrub, 2022-10-30:
[image]

Changed to winter time, it is 5AM with the new time. The graph has prediction lines starting in the past...

Alamo04, 2022-10-30:
@robertrub
Without Logs, they can't say anything about that.
They also don't know your Phone, your pump, your AAPS version....

So far I hear from our german community, they didn't have this issue. But I think they are all sleeping at this time. :)
Did that issue still exist, or was it self fixing ?

robertrub, 2022-10-30:
It was "fixed" when I got up around 8AM. 

It is probably just a graphic glitch and happens once a year, for winter time change ;)

robertrub, 2022-10-31:
There are other glitches too with DST. Nothing serious though.

[image]

closed 2023-08-22 by robertrub