nightscout/AndroidAPS #2146: DST causes double data in statistics in V3.1.0.3 dev-c

koelewij, 2022-10-31:
After DST, Statistics shows double data lines. 
[image]

Though it seems not that serious, functions like Dynamic ISF are using those statistics and might be influenced by this behavior. 
It would be better that the day involved would be an hour longer/shorter instead of generating a separate entry with unexplainable data.

robertrub, 2022-10-31:
Same here

[image]

MilosKozak, 2022-10-31:
cause by daylight saving change when day is longer than 24h

closed 2022-10-31 by MilosKozak