nightscout/AndroidAPS #2148: Request: Split bolus as bolus + ecarbs

NiHoYo, 2022-10-31:
Hey!

As far as I am aware, this feature is not implemented and I couldnt find a similar request. I think it would be of great help (at least for how I use AAPS).

My "workflow" when eating is to enter a bolus and then enter E-Carbs seperately, which the SMB will deal with.

The thing is, for almost all foods you would need to enter E-Carbs when using SMB. At least for me, it doesnt work without. You need to enter bolus and E-Carbs seperately, in a different menu. Its inconvenient and therefor I often forget to enter E-Carbs.

It would be nice to be able to input a direct bolus and E-Carbs at the same time, much like a split bolus.
Maybe this function could be activated in settings, just like the SuperBolus, and shown as a seperate button at the bottom. 

If would be very straightforward and only needs 4 inputs:
- amount of carbs/KH/BE
- % of Carbs/BE/KH/Bolus to be given directly
- Timespan of E-Carbs
- Delay of bolus

Here are a few examples: 
- You eat Pizza and bolus 40g carbs after 15 minutes. 40 E-Carbs will be input over 4 hours. Your input: 80g carbs, 50% direct, 4h timespan, 15min delay
- You eat sugar and bolus 24 carbs after 0 minutes, 0 E-Carbs will be input over 0 hours. Input: 24g carbs, 100% direct
- You eat a chicken filet and bolus 0 carbs after 0 minutes. 12 E-Carbs will be input over 4 hours. Input: 12g carbs, 0% direct, 4h delay.

On top of that, the input could easily be done with 4 sliders and the distribution could be shown as a very simple graph.

What do you think of the idea? I could not be of much help with the implementation, I think. I could refine the concept and write some java/kotlin.

Thanks a lot :) Nils

Andries-Smit, 2022-10-31:
Issue overlaps with #1952 ? Would be good if it can be combined with a food database #2076 

@NiHoYo Looking at your example, you might have interest in Fat Protein Units (FPU) calculation
https://www.ncbi.nlm.nih.gov/pmc/articles/PMC2901033/ 
More explaining https://www.diabetesdaily.com/learn-about-diabetes/treatment/insulin-101/how-to-use-insulin/how-to-calculate-bolus-insulin-dosing-for-protein/

An app that calculates the units https://rueth.info/index.php/en/easyfpu/easyfpu-for-ios

NiHoYo, 2022-10-31:
Thanks a lot Andries :) 

I have seen the other Issues and its very similar to #1952. Just that I would like E-Carbs as the second split half...

Adding a food database would be an awesome feature. But in the context of this issue, it might add too much complexity for its own good. 

Its not about **what** to input, but **how**.  I am hoping for a menu that allows me to da a bolus/e-carb split with a few clicks. In order to give the correct input, I of course need to know about FPE first.

naromero77, 2023-06-09:
I agree that would be very useful. Especially for young kids with short attention span.

ChristianeVie, 2024-03-04:
I also agree, that this would make life easier. Also fits to my feature request 3255...same direction...

MilosKozak, 2026-10-15:
if v4 you can define food type

closed 2026-10-15 by MilosKozak