nightscout/AndroidAPS #2154: Stats in AAPS inaccurate (perhaps)  -  Dev AIMI

ShaggydogUK, 2022-11-02:
Reporting bugs
--------------
- Time noted is: 
-  Build version 3.1.0.3 -dev-b-AIMI
the following screens are shown under pump stats:  
[image]
and AAPS stats.
[image]
The AAPS stats are showing 2 lines for 30/11 and many of the values from the pump stats do not agree with the AAPS stats. There been 3 cannula changes during this time, each one using a 0.4 unit prime that is done from the pump while the pump is disconnected.

Els3401, 2022-11-03:
I have the same problem. But I have a combo, and dev c aimi. Bolus in statistics is 0. 30/10 is double.
[image]
[image]
[image]
[image]

rosseland, 2022-11-05:
Same issue here. Rebuilt on c80e29f626cbfe40d2a980b84e2f47bc0d520423
Also, statistics are different on aaps client(follower phone) and in aaps. (none are correct) So, might there be some caching thats not updated?
And afaik, Dyn-ISF uses these stats, which now will not get the correct TDD

thebarefootboy, 2022-11-05:
Definitely something very strange happening with my stats too. Noticed this has happened since the UK summertime ended last weekend (since the double entries for 30 Oct).

None of the stats for bolus or carbs are accurate.

Very little bolus is recorded - showing only 0-2iu instead of ~20iu per day , same for carbs only recording 15-30g instead of 150-180g per day. Basal appears to be ok.

The correct amounts of insulin and carbs are uploaded to Nightscout.

I did a database reset and thought the issue was fixed, as while the data is in the "Today" section the data remains accurate but as soon as the next day starts the accurate data is lost.

Using version 3.1.0.3-dev-c (rebuilt after the summertime change). Also using Dynamic-ISF so am concerned that it is not getting accurate TDD, so have disabled it for now.

rosseland, 2022-11-07:
Its now been over a week since the double entry 30/10, so its out of scope for the TDD calulation. Numbers are now correct, both in AAPS and client.

thebarefootboy, 2022-11-07:
I can confirm in my case as soon as the TDD Statistics section stopped showing '8 days' (ie: the 2 entries for end of BST on 30 Oct rolled over) all the days since then that were showing incomplete/incorrect data are now showing the correct data.

So, all the daily  insulin/carb entries stored in the database are correct but there is definitely an issue that caused the display of new daily (bolus and carbs) data to be shown incorrectly on the Statistics page during the week where the 'extra' day appeared in the stats and the data was summarising 8 days.

MilosKozak, 2022-11-30:
improved by https://github.com/nightscout/AndroidAPS/commit/0ff8161cda29477d2c179332702d6d28948febf0

closed 2022-11-30 by MilosKozak