nightscout/AndroidAPS #2165: Loop not executing when BG is changing version 3.1.0.3

margogoodfellow, 2022-11-05:
Timing 7:21 it was last time when loop has executed. Bg has changed again but loop hasn't executed again. Please see pictures.  I have done bew build, passed all objectives and opened new nightscout.  Also done this on new phone.  
Please can you look into this and this is happening to me all the time in every version apart of 3.0.0

Link to logs : https://drive.google.com/file/d/1YHQ2yB-khQ0USeA-kuraAKfCV35tLObT/view?usp=drivesdk

Time 7:21-7.29

> 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

MilosKozak, 2022-11-30:
In logs i see AMA executing regulary even after 7:21

margogoodfellow, 2022-12-02:
I don't know then what's going on.  As you can see on the pictures,  loop
wasn't executing.

MilosKozak, 2022-12-04:
added more logging https://github.com/nightscout/AndroidAPS/commit/bd35007f44786235788b9879ffae5de73149f4c2, provide new logs pls

MilosKozak, 2022-12-04:
duplicate to https://github.com/nightscout/AndroidAPS/issues/2130

closed 2022-12-04 by MilosKozak