nightscout/AndroidAPS #2168: Wear: blank AAPS watchface on Samsung Watch 4

dexdan, 2022-11-06:
Have problem getting AAPS watch face on Samsung Watch 4 with the latest AAPS version (3.1.0.3-dev-c-AIMI). I got a new watch, and compiled / installed the wear apk fine via adb. (ie, both phone and watch having matching latest builds). The wear app itself worked OK (I could bolus from the watch, albeit couldn’t get into settings) and complications worked on other watch faces – but AAPS watchfaces just showed totally black. Tried uninstalling / reinstalling and watch/phone reboots.  

My workaround for now is to use a wear apk compiled (dev c, but not 100% sure) in September (version 27D2709646-2022.09.08 as per watch settings screen) which seems to work OK (albeit being misaligned with the version running on the phone). 

So it seems there has been a change to dev aaps wear code since September which has broken watch faces for Samsung Watch 4.

koelewij, 2022-11-06:
This seems to have happened somewhere between October 25th and November 1st. Installing a wear build of October 25th solved the issue.

Danielkharkiv, 2022-11-08:
I have the same problem. Could you help and show how you have chosen the date of the wear build?

koelewij, 2022-11-08:
I had an old version as backup. I don’t really know how to do this in AS. You might ask on discord. You can also try the master build. Possibly that works fine with the dev version on the phone. No major changes I believe.

Danielkharkiv, 2022-11-08:
I was trying to do build wear apk in master dev  as well, but it was the still same problem.

Danielkharkiv, 2022-11-08:
It looks like the best way is to wait a new commit

Andries-Smit, 2022-11-11:
It seems that something is wrong with the line chart. So, until it is fixed you can take a watch face without a chart, that will work
> FATAL EXCEPTION: main
                                                                                                    Process: info.nightscout.androidaps, PID: 25711
                                                                                                    java.lang.RuntimeException: Unable to create service info.nightscout.androidaps.watchfaces.AapsWatchface: android.view.InflateException: Binary XML file line #162 in info.nightscout.androidaps:layout/activity_home: Binary XML file line #162 in info.nightscout.androidaps:layout/activity_home: Error inflating class lecho.lib.hellocharts.view.LineChartView
                                                                                                    	at android.app.ActivityThread.handleCreateService(ActivityThread.java:4215)

Andries-Smit, 2022-11-12:
Issue is caused be depended libraries that are old, to make them work on modern Android, jettifier could be used. However this was disabled in commit 
https://github.com/nightscout/AndroidAPS/commit/e7c41beab74306ee3b5c68c41aac9bf05da36034

Probable reason, it slows the build. I could recommend "pre-jetifier" the decencies as described here.
https://adambennett.dev/2020/08/disabling-jetifier/

For all inpatient, that do not want to way on a fix, just enable jetifier again, and rebuild

closed 2022-11-13 by MilosKozak

dexdan, 2022-11-13:
Many thanks both!

Danielkharkiv, 2022-11-13:
Thanks, everybody for sharing your thoughts and your propositions. The new update solved the problem. Thanks to developers