nightscout/AndroidAPS #2170: AAPS stopped receiving BG data from BYODA

lydiaengler, 2022-11-06:
version 3.1.0.3

Build1de236a602-2022.0
8.09-10:42

Full release
NS 14.0.6

System working fine for 2 weeks.

Suddenly today it stopped receiving any BG data from BYODA., but BYODA is working fine.

Instead of a BG reading it just displayed the old one with (180 mins ago after 3 hours), then after the reinstallation it just says n/a ??

Also, the app is crashing regularly - I think when it tries to receive a reading from BYODA.

Have tried reinstalling the app, including after clearing cache and all data.

Log files attached

As I said, this was working fine until today. We changed nothing.  It simply stopped receiving BG readings.

On this phone also, it workled flawlessly for a year on an earlier version. I only changed phone 2 weeks ago (for a bigger screen), the phone I was on before did the same thing - worked OK for a few weeks then just failed one day. I thought it was somethinhg to with too little RAM but now it has happened on the old phone I have rehabilitated, I don't think that had anything to do with it.

[AndroidAPS_LOG_1667764675517.log.zip](https://github.com/nightscout/AndroidAPS/files/9946797/AndroidAPS_LOG_1667764675517.log.zip)

lydiaengler, 2022-11-07:
I have an update.  I reinstalled BYODA and that seemed to resolve it.  I then read that the problem was my NS version was too old, I have now updated it.

Is it possible that the NS version being too old could result in it working fine for a few weeks, then suddenly stopping working?

Alamo04, 2022-11-07:
No.

AAPS works as a stand alone system. NS is only needed for your reports and to finish the objectives.

lydiaengler, 2022-11-08:
Thanks.  I did actually try rebooting the phone to no avail.  It was only reinstalling BYODA which solved it.

lydiaengler, 2022-11-08:
Which onbiously begs the question of whether the issue was BYODA or AAPS.  The fault manifested only in AAPS but it was the reinstallation of BYODA which fixed it.

I have been in touch with the developer of BYODA via his Reddit page, but he cannot think of anything which could have caused this issue.

MilosKozak, 2022-11-15:
no idea as well

closed 2022-11-15 by MilosKozak