nightscout/AndroidAPS #2172: Last Dev-C : lost connection to watch face

robertrub, 2022-11-07:
[image]

@MilosKozak I lost connection to watch face after updating the phone apk and watch apk the the last dev-c.

New apk on phone and 3 days  "old" apk on watch did not communicate either.

overfrenk, 2022-11-07:
Same for me! Samsung watch4

robertrub, 2022-11-08:
To be more precise, AAPS Wear menu has the correct watch name (thus is connected) but BGs COB and IOB ate stale, not updated (even when I hit send resend dll data to watch.

robertrub, 2022-11-08:
[image]

This version does not solve the connection problem to the watch.

Andries-Smit, 2022-11-11:
Duplicate of #2168

Andries-Smit, 2022-11-12:
Issue is caused be depended libraries that are old, to make them work on modern Android, jettifier could be used. However this was disabled in commit https://github.com/nightscout/AndroidAPS/commit/e7c41beab74306ee3b5c68c41aac9bf05da36034

(Also the preference menu is impacted)

Probable reason, it slows the build. I could recommend "pre-jetifier" the decencies as described here.
https://adambennett.dev/2020/08/disabling-jetifier/

For all inpatient, that do not want to way on a fix, just enable jetifier again, and rebuild

closed 2022-11-13 by MilosKozak

robertrub, 2022-11-13:
Hi @Andries-Smit 

You last modifications did the job. The watchface is working again. Tnx a lot. 

Robby

PR #2199 (2022-11-13): wear: fix jetifier aar libs
Wear OS preference menus and watch faces with charts are depended on old stuff and required Jetifier, when it was disabled https://github.com/nightscout/AndroidAPS/commit/e7c41beab74306ee3b5c68c41aac9bf05da36034 , it causes trouble, this PR fixes #2172 and #2168

- Moved hellocharts to AAR lib (Sourced from https://search.maven.org/artifact/com.github.lecho/hellocharts-library/1.5.8/jar)
- User jetifier standalone to fixup the AAR lib files https://developer.android.com/studio/command-line/jetifier