nightscout/AndroidAPS #2207: Future request

overfrenk, 2022-11-16:
Future request
--------------
Is it possible to add the amount of insulin injected into the site to the state of the cannula in addition to the time? Obviously resettable when the cannula is changed. To also get an idea of how much insulin was made at the site.

overfrenk, 2022-11-21:
@Andries-Smit My daily tdd is very variable and I never completely change the cannula and reservoir. So having 290U available with a full tank and consuming it in 2 or 4 days it would be very useful for me to know how much insulin I dispensed in the site, because beyond 120/150U or within 2/3 days the site starts to hurt and the action insulin is no longer good. The value would be convenient for me just below the cannula time or to the left of the cannula icon.

Andries-Smit, 2022-11-21:
@overfrenk The canula information is shown in the HOME and the ACTION page. The limited space on the home page statuslight bar will make it hard to place it there. Will be good enough to place it only ACTION page?
[image]

overfrenk, 2022-11-21:
@Andries-Smit PERFECT!

Andries-Smit, 2022-11-24:
@overfrenk The feature is implemented and is available in the dev branch or in the next release

overfrenk, 2022-11-24:
@Andries-Smit Thank you! I tried to generate the dev build, like almost every day, but crash...... Waiting for a fix

closed 2022-11-24 by MilosKozak