nightscout/AndroidAPS #2208: [DEV] aaps cannot load the settings and restarts after update to latest dev

ledrexter, 2022-11-16:
Hi,
I with Xiaomi 11T Pro **Android 12** SKQ1.211006.001  MIUI 13.0.7.
The pump is DanaRv2.
I build the dev branch from today and when I update aaps on the phone it crashes very fast (less than 7-8 seconds).
I tried forcing aaps exit, clear the cash only and restart but it does not help.
I tried clearing all data, when I start the app the inti process is working fine. It requests for the permissions, admin passwords and it asks for setting file to restore. When I select a setting the aaps crashes again. I tried latest setting file, as well as some pretty old setting I have. It is still the same.

Only once I was fast enough, to change the settings to virtual pump and then the crashes stopped. When I set back DanaRv2 it cashed again.

A simple update to AndroidAPS 3.1.0.2 is working fine and the issue is gone. I tried 4 different days in the last two weeks to pull the dev to up-to-date and rebuild but the result is always the same. 
The last dev build I used without this issue was 3.1.03-dev-b from 4th of October.

In the [AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/10020213/AndroidAPS.log)file I see the following:
11:14:52.144 [main] D/PUMPCOMM: [MsgSetUserOptions.<init>():15]: **NO USER OPTIONS LOADED EXITING**!

A friend of mine with the **same phone** and pump, but his **DanaR is not version 2** and he gets the same issue when he try to build and update to dev.

I made a [video ](https://drive.google.com/file/d/1N7v4soDHMlrk9YzpHMqifUJ2lHTjFLOf/view?usp=sharing) as well in case it helped.

Please let me know if I can assist in furthered investigation the issue.

Kind regards,
Petar

ledrexter, 2022-11-16:
For some reason the video become 360p here is a [link ](https://youtu.be/qzlBD09ipEQ) to the video with better quality.

MilosKozak, 2022-11-16:
i see crashes related to missing BT permission

MilosKozak, 2022-11-30:
there is a fix fo latest MIUI in dev
test again

closed 2022-12-04 by MilosKozak

ledrexter, 2022-12-21:
> there is a fix fo latest MIUI in dev test again

The issue is resolved now.
Thanks for the hard support!