nightscout/AndroidAPS #2213: aaps promted rebuild when installed on new device, then did not promp when reinstalled on old device. Now 60 days have passed and aaps has not reverted to LGS

trinawashere, 2022-11-23:
- Issue occurred September 11/2022 8:13pm PST
- was having phone issues so installed aaps onto a new device temporarily, then uninstalled aaps on old device, then reinstalled aaps on old device when the device was fixed
- when I installed aaps on the new device I got a prompt to rebuild: "New version for at least 174 days available! Fallback to LGS after 60 days, loop will be disabled after 90 days"
-  when I switched back to the old device and reinstalled aaps the notification did not come again 
- it has now been 72 days and aaps has not fallen back to LGS
-
Build version 3.0.0.1

[image]
[AndroidAPS._2022-11-19_00-00-05_.1.zip](https://github.com/nightscout/AndroidAPS/files/10073464/AndroidAPS._2022-11-19_00-00-05_.1.zip)
[AndroidAPS._2022-11-19_00-00-05_.2.zip](https://github.com/nightscout/AndroidAPS/files/10073465/AndroidAPS._2022-11-19_00-00-05_.2.zip)
[AndroidAPS._2022-11-19_00-00-05_.3.zip](https://github.com/nightscout/AndroidAPS/files/10073466/AndroidAPS._2022-11-19_00-00-05_.3.zip)
[AndroidAPS._2022-11-19_00-00-05_.4.zip](https://github.com/nightscout/AndroidAPS/files/10073467/AndroidAPS._2022-11-19_00-00-05_.4.zip)
[AndroidAPS._2022-11-19_00-00-05_.5.zip](https://github.com/nightscout/AndroidAPS/files/10073468/AndroidAPS._2022-11-19_00-00-05_.5.zip)
[AndroidAPS._2022-11-19_00-00-05_.6.zip](https://github.com/nightscout/AndroidAPS/files/10073469/AndroidAPS._2022-11-19_00-00-05_.6.zip)
[AndroidAPS._2022-11-19_00-00-05_.7.zip](https://github.com/nightscout/AndroidAPS/files/10073470/AndroidAPS._2022-11-19_00-00-05_.7.zip)
[AndroidAPS._2022-11-20_00-00-01_.0.zip](https://github.com/nightscout/AndroidAPS/files/10073471/AndroidAPS._2022-11-20_00-00-01_.0.zip)
[AndroidAPS._2022-11-20_00-00-01_.1.zip](https://github.com/nightscout/AndroidAPS/files/10073472/AndroidAPS._2022-11-20_00-00-01_.1.zip)
[AndroidAPS._2022-11-20_00-00-01_.2.zip](https://github.com/nightscout/AndroidAPS/files/10073473/AndroidAPS._2022-11-20_00-00-01_.2.zip)
[AndroidAPS._2022-11-20_00-00-01_.3.zip](https://github.com/nightscout/AndroidAPS/files/10073474/AndroidAPS._2022-11-20_00-00-01_.3.zip)
[AndroidAPS._2022-11-20_00-00-01_.4.zip](https://github.com/nightscout/AndroidAPS/files/10073475/AndroidAPS._2022-11-20_00-00-01_.4.zip)
[AndroidAPS._2022-11-20_00-00-01_.5.zip](https://github.com/nightscout/AndroidAPS/files/10073476/AndroidAPS._2022-11-20_00-00-01_.5.zip)
[AndroidAPS._2022-11-20_00-00-01_.6.zip](https://github.com/nightscout/AndroidAPS/files/10073477/AndroidAPS._2022-11-20_00-00-01_.6.zip)
[AndroidAPS._2022-11-21_00-00-00_.0.zip](https://github.com/nightscout/AndroidAPS/files/10073478/AndroidAPS._2022-11-21_00-00-00_.0.zip)
[AndroidAPS._2022-11-21_00-00-00_.1.zip](https://github.com/nightscout/AndroidAPS/files/10073479/AndroidAPS._2022-11-21_00-00-00_.1.zip)
[AndroidAPS._2022-11-21_00-00-00_.2.zip](https://github.com/nightscout/AndroidAPS/files/10073480/AndroidAPS._2022-11-21_00-00-00_.2.zip)
[AndroidAPS._2022-11-21_00-00-00_.3.zip](https://github.com/nightscout/AndroidAPS/files/10073481/AndroidAPS._2022-11-21_00-00-00_.3.zip)
[AndroidAPS._2022-11-21_00-00-00_.4.zip](https://github.com/nightscout/AndroidAPS/files/10073482/AndroidAPS._2022-11-21_00-00-00_.4.zip)
[AndroidAPS._2022-11-21_00-00-00_.5.zip](https://github.com/nightscout/AndroidAPS/files/10073483/AndroidAPS._2022-11-21_00-00-00_.5.zip)
[AndroidAPS._2022-11-21_00-00-00_.6.zip](https://github.com/nightscout/AndroidAPS/files/10073484/AndroidAPS._2022-11-21_00-00-00_.6.zip)
[AndroidAPS._2022-11-21_00-00-00_.7.zip](https://github.com/nightscout/AndroidAPS/files/10073485/AndroidAPS._2022-11-21_00-00-00_.7.zip)
[AndroidAPS._2022-11-22_00-00-02_.0.zip](https://github.com/nightscout/AndroidAPS/files/10073486/AndroidAPS._2022-11-22_00-00-02_.0.zip)
[AndroidAPS._2022-11-22_00-00-02_.1.zip](https://github.com/nightscout/AndroidAPS/files/10073487/AndroidAPS._2022-11-22_00-00-02_.1.zip)
[AndroidAPS._2022-11-22_00-00-02_.2.zip](https://github.com/nightscout/AndroidAPS/files/10073488/AndroidAPS._2022-11-22_00-00-02_.2.zip)
[AndroidAPS._2022-11-22_00-00-02_.3.zip](https://github.com/nightscout/AndroidAPS/files/10073489/AndroidAPS._2022-11-22_00-00-02_.3.zip)
[AndroidAPS._2022-11-22_00-00-02_.4.zip](https://github.com/nightscout/AndroidAPS/files/10073490/AndroidAPS._2022-11-22_00-00-02_.4.zip)
[AndroidAPS._2022-11-22_00-00-02_.5.zip](https://github.com/nightscout/AndroidAPS/files/10073491/AndroidAPS._2022-11-22_00-00-02_.5.zip)
[AndroidAPS._2022-11-22_00-00-02_.6.zip](https://github.com/nightscout/AndroidAPS/files/10073492/AndroidAPS._2022-11-22_00-00-02_.6.zip)

Philoul, 2022-12-06:
Latest version of AAPS is 3.1.0.3 so you have to update to latest release version (it's not a bug, it works as designed)
If you didn't get the alert on the other device, it was probably due to connection issue to internet...

MilosKozak, 2023-08-17:
https://github.com/nightscout/AndroidAPS/issues/2397

closed 2023-08-17 by MilosKozak