nightscout/AndroidAPS #2222: [Bug] [Combov2] [Latest dev] Preferences tab craches on Android 12

swissalpine, 2022-11-26:
Unfortunatly the preferences tab crashes on Android 12 (Samsung Galaxy Flip 4) and I can't pair a combo. Other preference screens on other tabs works fine. And there is no problem on my Jelly 2 with Android 11 with the same build. This problem also exists with old builds, which I also tried today: Android 11 -> no problem, Android 12 -> crash.
Here is the logcat:
```
2022-11-26 19:48:31.174 24888-24888/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: info.nightscout.androidaps, PID: 24888
    java.lang.ClassCastException: java.lang.String cannot be cast to java.lang.Integer
        at android.app.SharedPreferencesImpl.getInt(SharedPreferencesImpl.java:321)
        at androidx.preference.Preference.getPersistedInt(Preference.java:1813)
        at androidx.preference.SeekBarPreference.onSetInitialValue(SeekBarPreference.java:207)
        at androidx.preference.Preference.onSetInitialValue(Preference.java:1634)
        at androidx.preference.Preference.dispatchSetInitialValue(Preference.java:1607)
        at androidx.preference.Preference.onAttachedToHierarchy(Preference.java:1330)
        at androidx.preference.Preference.onAttachedToHierarchy(Preference.java:1345)
        at androidx.preference.PreferenceGroup.addPreference(PreferenceGroup.java:253)
        at androidx.preference.PreferenceGroup.addItemFromInflater(PreferenceGroup.java:173)
        at androidx.preference.PreferenceInflater.rInflate(PreferenceInflater.java:350)
        at androidx.preference.PreferenceInflater.rInflate(PreferenceInflater.java:351)
        at androidx.preference.PreferenceInflater.inflate(PreferenceInflater.java:161)
        at androidx.preference.PreferenceInflater.inflate(PreferenceInflater.java:112)
        at androidx.preference.PreferenceManager.inflateFromResource(PreferenceManager.java:217)
        at info.nightscout.androidaps.activities.MyPreferenceFragment.addPreferencesFromResource(MyPreferenceFragment.kt:298)
        at info.nightscout.androidaps.activities.MyPreferenceFragment.onCreatePreferences(MyPreferenceFragment.kt:181)
        at androidx.preference.PreferenceFragmentCompat.onCreate(PreferenceFragmentCompat.java:161)
        at info.nightscout.androidaps.activities.MyPreferenceFragment.onCreate(MyPreferenceFragment.kt:163)
        at androidx.fragment.app.Fragment.performCreate(Fragment.java:3090)
        at androidx.fragment.app.FragmentStateManager.create(FragmentStateManager.java:475)
        at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:257)
        at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:1899)
        at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:1817)
        at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:1760)
        at androidx.fragment.app.FragmentManager.dispatchStateChange(FragmentManager.java:2985)
        at androidx.fragment.app.FragmentManager.dispatchActivityCreated(FragmentManager.java:2895)
        at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:263)
        at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:351)
        at androidx.appcompat.app.AppCompatActivity.onStart(AppCompatActivity.java:248)
        at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1473)
        at android.app.Activity.performStart(Activity.java:8351)
        at android.app.ActivityThread.handleStartActivity(ActivityThread.java:4262)
        at android.app.servertransaction.TransactionExecutor.performLifecycleSequence(TransactionExecutor.java:221)
        at android.app.servertransaction.TransactionExecutor.cycleToPath(TransactionExecutor.java:201)
        at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:173)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2498)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loopOnce(Looper.java:226)
        at android.os.Looper.loop(Looper.java:313)
        at android.app.ActivityThread.main(ActivityThread.java:8855)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)
        ```

BUILD: 188279b616-2022.11.26

clvsjr9, 2022-11-26:
Android 10 crashes too

MilosKozak, 2022-11-26:
which preference screen?

MilosKozak, 2022-11-26:
no issue with A12 on my test phone

clvsjr9, 2022-11-26:
Android 10 (Samsung Galaxy S9). Combov2 Preferences and Preferences with General, Overview, Treatments safety, etc crashes.

swissalpine, 2022-11-26:
Android 12 (Samsung Flip 4) with Combov2 Preferences, openAPS Preferences crashes f. e., other works (Free Peak Oref, Loop f. e.). As written above I see no issues on my Jelly 2.

MilosKozak, 2022-11-26:
hmmm. no idea. could you test which commit is responsible?

clvsjr9, 2022-11-26:
git log -1  

commit 7fae8335a66a2031039dedc655079a38327a4a31 (HEAD -> dev, origin/dev)
Merge: e47cdded0b d4c6759dd8
Author: Milos Kozak <m.kozak@sysop.cz>
Date:   Sat Nov 26 00:22:00 2022 +0100

    Merge branch 'dev' of https://github.com/nightscout/AndroidAPS into dev

swissalpine, 2022-11-27:
The combov2 preferences don't seem to work with Android 12 (Samsung), since my first merge on 06.11.22.
At least the openAPS SMB preferences and Overview preferences work up to
BUILD cc25d1b967 on 25.11.22, after that no more.
As I said, everything is fine on the Jelly with Android 11.

MilosKozak, 2022-11-27:
could you test if it works on clean installation?

swissalpine, 2022-11-27:
It was a "clean installation". I got the phone yesterday ...
Or do you mean an installation without own changes?
I did the same with today's dev without any personal change and complete reinstallation. Same result:
Overview > 3 point menu > Preferences: Crash
Combo v2 > Preferences: Crash
openAPS SMB > Preferences: now no crash
In the meantime, the Samsung Flip4 pulled Android 13, which brought no change at all.

clvsjr9, 2022-11-27:
How should I find the commit number equivalent to BUILD cc25d1b967 on 25.11.22 and try to do a git hard reset for it and evaluate on Android 10 to see if it works? I can do a clean installation too.

swissalpine, 2022-11-27:
Last commit for my build with working pref for openAPS SMB and Overview was:
2e034e91fabae11c55f4aac59458ad3351442f27
Revert "resolve deprecation warning"

MilosKozak, 2022-11-27:
> It was a "clean installation". I got the phone yesterday ... Or do you mean an installation without own changes? I did the same with today's dev without any personal change and complete [...]

I mean without imported data

swissalpine, 2022-11-27:
Ok. I just tried this, but I can't complete the setup wizard because I can't get past the screen pump without crashing. With the old Combo driver nothing happens, with Virtual Pump and clicking Next the setup dialogue crashes.

swissalpine, 2022-11-27:
Second attempt: This time with location authorisation for bluetooth and it worked ...
I installed "clean" without importing the settings - and, tata, no more crashes.
But: Can you fix this, @Milos. Because this way I lose not only my settings, but also my objectives.

clvsjr9, 2022-11-27:
I did a "clean" install with BUILD cc25d1b967 on 25.11.22 (from @swissalpine ) on Android 10 and 100% working now. I will try to do with current dev version with "clean" too...

clvsjr9, 2022-11-27:
With latest dev version 100% OK on Android 10 with "clean install". No more crashed with Preferences access and ComboV2 Preferences too. When I imported Settings the problem appears again.

MilosKozak, 2022-11-27:
fixed by https://github.com/nightscout/AndroidAPS/commit/57c96770b06aceb6d7868a923bc7e30d6032d88f

clvsjr9, 2022-11-27:
@MilosKozak very thanks !!!

MilosKozak, 2022-11-27:
working?

clvsjr9, 2022-11-27:
on my Android 10 yes.

swissalpine, 2022-11-27:
Same here: Fixed for Android 13. Thank you! @MilosKozak

closed 2022-11-28 by swissalpine