nightscout/AndroidAPS #2223: [DEV] Crash loop when xDrip battery status of G6 is available in NS

jbr7rr, 2022-11-27:
Latest dev 305395758f-2022.11.27 (dev-c-nscv3)

Problem occurred at: 2022-11-27 09:49:59.450

Problem description:
AAPS comes in a crash loop, seems to be caused by battery status from xDrip in Nightscout.
See this line in the stacktrace:
" java.lang.NumberFormatException: For input string: "6 days (voltage: 308/289)" "

Context: 
Switched from Tsunami to latest dev, cleaned everything tried to pull data from NS to fill in latest day. after activating the NSClient plugin AAPS came into a crash loop. 
After a search on discord I came onto this post (user does not seem to report the issue here yet :( ): 
https://discord.com/channels/629952586895851530/629954570394533889/1041885298591797308

After lots of searching in MongoDB I got frustrated and just deleted the devicestatus collection. This seems to be a viable workaround for now.

See attachment for logs of the accident and a stacktrace.
[logs_and_stacktrace.zip](https://github.com/nightscout/AndroidAPS/files/10097987/logs_and_stacktrace.zip)

MilosKozak, 2022-11-27:
I need the json with data

jbr7rr, 2022-11-27:
> I need the json with data

The JSON of the DB entry I assume?

dv1, 2022-11-27:
@MilosKozak 

Example:

```
      {
         "_id" : "63828b92712ff2fa64f1ed6b",
         "created_at" : "2022-11-26T21:56:34.861Z",
         "device" : "G6 Transmitter",
         "mills" : 1669499794861,
         "uploader" : {
            "battery" : "38 days (voltage: 307/288)",
            "days" : 38,
            "daysEstimate" : "38 / 38.2",
            "lastQueried" : 1669499789395,
            "resistance" : 0,
            "resistance_status" : "GOOD",
            "status" : "OK",
            "temperature" : 27,
            "type" : "DEXCOM_TRANSMITTER",
            "voltagea" : 307,
            "voltageb" : 288
         },
         "utcOffset" : 60
      },
```

The bug is [here in xDrip](https://github.com/NightscoutFoundation/xDrip/blob/master/app/src/main/java/com/eveningoutpost/dexdrip/UtilityModels/NightscoutBatteryDevice.java#L126), caused by [this commit](https://github.com/NightscoutFoundation/xDrip/commit/92956f73f49dc199e2539ba055ec0d647876b91e).

jbr7rr, 2022-11-27:
Was able te reproduce it. by re-enabling the upload battery and forcing it by a small change in the xDrip code.

This data triggers it:

`{"_id":{"$oid":"63836bd8bfb8b72dc9a41dd1"},"device":"G6 Transmitter","uploader":{"days":{"$numberInt":"7"},"daysEstimate":"7 / 7.2","status":"OK","voltagea":{"$numberInt":"308"},"voltageb":{"$numberInt":"288"},"resistance":{"$numberInt":"0"},"resistance_status":"GOOD","temperature":{"$numberInt":"28"},"battery":"7 days (voltage: 308/288)","lastQueried":{"$numberDouble":"1.6695572056900E+12"},"type":"DEXCOM_TRANSMITTER"},"created_at":"2022-11-27T13:53:28.906Z","utcOffset":{"$numberInt":"0"}}`

dv1, 2022-11-27:
@MilosKozak See the commit above I just uploaded to my fork. (It replaces my previous hack.) Do you approve, or do you see problems with it? If you are OK with it, I'll turn it into a PR.

jbr7rr, 2022-11-28:
Thanks @dv1
Can confirm your fix works like a charm, reproduced it with old build by adding the data to the database. Then flashed new build. no more crashes!