nightscout/AndroidAPS #2230: [DEV} Dynamic ISF autosens ratio does not respect maximum in settings

jbr7rr, 2022-11-28:
Occurs on the following builds:
dev 305395758f-2022.11.27 (dev-c-nscv3)
dev 5fa7fb34d4-2022.11.28 (dev-c-nscv3)

When enabling this function in the dynamicISF plugin:
[image]

There is a sensitivity multiplier enabled, which can multiply the basal that is used in calculations uncapped. this can result in unwanted results:
[image]

After some digging and discusions in discord, this is the code that is responsible for it
The ratio is calculated here:
https://github.com/nightscout/AndroidAPS/blob/a79053e5e2ebc7f70b5657ec8029439423c58a55/plugins/aps/src/main/java/info/nightscout/plugins/aps/openAPSSMBDynamicISF/DetermineBasalAdapterSMBDynamicISFJS.kt#L308
Is assigned in determine_basal.js here:
https://github.com/nightscout/AndroidAPS/blob/a79053e5e2ebc7f70b5657ec8029439423c58a55/plugins/aps/src/main/assets/OpenAPSSMBDynamicISF/determine-basal.js#L235
And is used to set basal here:
https://github.com/nightscout/AndroidAPS/blob/a79053e5e2ebc7f70b5657ec8029439423c58a55/plugins/aps/src/main/assets/OpenAPSSMBDynamicISF/determine-basal.js#L242

In master this calculation seems to be capped to the autosens max setting:
https://github.com/nightscout/AndroidAPS/blob/1de236a602d1bceae293e1e89f28d9e1f43432bd/app/src/main/assets/OpenAPSSMBDynamicISF/determine-basal.js#L331

MilosKozak, 2022-11-28:
limits mentioned are valid for autosens.
DynISF operates in much wider range and these limits are not applied

closed 2022-11-28 by MilosKozak

tim2000s, 2022-11-29:
Somewhere along the line, the autosens.max and autosens.min checks have been removed from the TDD based ratio. Looking back in the history, it looks like that may have been with the Kotlin conversion, as at the 1.6.5 version, lines 328 to 339 restricted the TDD based equivalent of Autosens to use the max and min provided by the user.