nightscout/AndroidAPS #2239: [EOPATCH2] unable to change profile

MarvelousMarvin, 2022-11-29:
Reporting bugs
--------------
- 10:40 hours on 29 November. I haven't been able to do a (temp) profile change
-
- AAPS 3.1.0.3-dev-c-nscv3
Build: 7fae8335a6-2022.11.26
Remote: https://github.com/nightscout/AndroidAPS.git
Flavor: fullrelease
Nightscout versie: 14.2.6
- 
[image]

MilosKozak, 2022-11-30:
logs?

MarvelousMarvin, 2022-11-30:
I did send them right after the incident by mail. I just got a msg that is wasn't delivered 24 hours later.

[image]

MarvelousMarvin, 2022-11-30:
I did send them by email right after the incident. They just got back to me.

[image]

I haven't figured out yet how to post them here.

MarvelousMarvin, 2022-11-30:
[AndroidAPS._2022-11-29_01-00-01_.18.zip](https://github.com/nightscout/AndroidAPS/files/10123157/AndroidAPS._2022-11-29_01-00-01_.18.zip)
[AndroidAPS._2022-11-29_01-00-01_.17.zip](https://github.com/nightscout/AndroidAPS/files/10123158/AndroidAPS._2022-11-29_01-00-01_.17.zip)
[AndroidAPS._2022-11-29_01-00-01_.16.zip](https://github.com/nightscout/AndroidAPS/files/10123159/AndroidAPS._2022-11-29_01-00-01_.16.zip)
[AndroidAPS._2022-11-29_01-00-01_.15.zip](https://github.com/nightscout/AndroidAPS/files/10123160/AndroidAPS._2022-11-29_01-00-01_.15.zip)
[AndroidAPS._2022-11-29_01-00-01_.14.zip](https://github.com/nightscout/AndroidAPS/files/10123161/AndroidAPS._2022-11-29_01-00-01_.14.zip)
[AndroidAPS._2022-11-29_01-00-01_.13.zip](https://github.com/nightscout/AndroidAPS/files/10123162/AndroidAPS._2022-11-29_01-00-01_.13.zip)
[AndroidAPS._2022-11-29_01-00-01_.12.zip](https://github.com/nightscout/AndroidAPS/files/10123163/AndroidAPS._2022-11-29_01-00-01_.12.zip)
[AndroidAPS._2022-11-29_01-00-01_.11.zip](https://github.com/nightscout/AndroidAPS/files/10123164/AndroidAPS._2022-11-29_01-00-01_.11.zip)
[AndroidAPS._2022-11-29_01-00-01_.10.zip](https://github.com/nightscout/AndroidAPS/files/10123165/AndroidAPS._2022-11-29_01-00-01_.10.zip)
[AndroidAPS._2022-11-29_01-00-01_.1.zip](https://github.com/nightscout/AndroidAPS/files/10123166/AndroidAPS._2022-11-29_01-00-01_.1.zip)
[AndroidAPS._2022-11-29_01-00-01_.0.zip](https://github.com/nightscout/AndroidAPS/files/10123167/AndroidAPS._2022-11-29_01-00-01_.0.zip)
[AndroidAPS._2022-11-28_16-35-24_.16.zip](https://github.com/nightscout/AndroidAPS/files/10123169/AndroidAPS._2022-11-28_16-35-24_.16.zip)
[AndroidAPS._2022-11-28_16-35-24_.15.zip](https://github.com/nightscout/AndroidAPS/files/10123170/AndroidAPS._2022-11-28_16-35-24_.15.zip)
[AndroidAPS._2022-11-29_01-00-01_.3.zip](https://github.com/nightscout/AndroidAPS/files/10123171/AndroidAPS._2022-11-29_01-00-01_.3.zip)
[AndroidAPS._2022-11-29_01-00-01_.2.zip](https://github.com/nightscout/AndroidAPS/files/10123172/AndroidAPS._2022-11-29_01-00-01_.2.zip)
[AndroidAPS._2022-11-29_01-00-01_.5.zip](https://github.com/nightscout/AndroidAPS/files/10123173/AndroidAPS._2022-11-29_01-00-01_.5.zip)
[AndroidAPS._2022-11-29_01-00-01_.4.zip](https://github.com/nightscout/AndroidAPS/files/10123174/AndroidAPS._2022-11-29_01-00-01_.4.zip)
[AndroidAPS._2022-11-29_01-00-01_.7.zip](https://github.com/nightscout/AndroidAPS/files/10123175/AndroidAPS._2022-11-29_01-00-01_.7.zip)
[AndroidAPS._2022-11-29_01-00-01_.6.zip](https://github.com/nightscout/AndroidAPS/files/10123176/AndroidAPS._2022-11-29_01-00-01_.6.zip)
[AndroidAPS._2022-11-29_01-00-01_.8.zip](https://github.com/nightscout/AndroidAPS/files/10123177/AndroidAPS._2022-11-29_01-00-01_.8.zip)
[AndroidAPS._2022-11-29_01-00-01_.9.zip](https://github.com/nightscout/AndroidAPS/files/10123178/AndroidAPS._2022-11-29_01-00-01_.9.zip)

Sorry, I have uploaded whole of 29 November. 🫣

MilosKozak, 2022-12-01:
there is only message in log that driver reported fail.
is it a reproducible error?

MilosKozak, 2022-12-01:
note:  log .5.zip

MarvelousMarvin, 2022-12-01:
Every time when I'm changing profile I get the same error.

[image]

MilosKozak, 2022-12-01:
turn on all PUMP logging

MarvelousMarvin, 2022-12-01:
I have tried to send a recent low by mail

Not sure if the correct logging is on. Couldn't figure out wich one was missing

jbr7rr, 2022-12-01:
I have the same problem with eoPatch, Build: ca06b948bf-2022.11.30
After intial pairing I was able to set the profile at different percentages. But after about 2 Hours The profile was unable to switch back from a 130% temp profile.

However after suspending the pump from the eopatch screen, en re-enabling from the same screen, the driver was able to set the profile again.

19:29 > 130% ends > failure
19:31 > Manual set > failure
19:32 > Manual set > failure
19:41 > Manual set > failure
19:44 > Manual set after workaround > success

See logs
[AndroidAPS._2022-12-01_Logs_jb.zip](https://github.com/nightscout/AndroidAPS/files/10135433/AndroidAPS._2022-12-01_Logs_jb.zip)

MarvelousMarvin, 2022-12-01:
Workaround does work for. That's a great start!

My best guess is that the pump doesn't accept profile change during active mode.

jbr7rr, 2022-12-02:
I added some additional logging and catched the error again.
The function setNewBasalProfile returns early at this line:
https://github.com/nightscout/AndroidAPS/blob/6172ca8d6391eae90ad273f2e7c1e7504bd78683/pump/eopatch/src/main/java/info/nightscout/androidaps/plugins/pump/eopatch/EopatchPumpPlugin.kt#L180

That this is triggered when a bolus is active I understand, but what needs to happen when a temp basal is active? I think temp basal should be cancalled and the new profile set?

Still figuring out how to implement that. Not that familier with Kotlin and AAPS code base :)

MilosKozak, 2022-12-04:
fixed by https://github.com/nightscout/AndroidAPS/commit/0c833d087ba0bdd57824cfa92a3fa28eb7d5db0c

closed 2022-12-04 by MilosKozak