nightscout/AndroidAPS #2250: **AndroidAPS** 3.1.0.3-dev-d showing logo on DASH History view (affects readability)

vanelsberg, 2022-12-03:
**AndroidAPS** 3.1.0.3-dev-d
**Build**: 6172ca8d
**Flavor**: fullrelease / Engineering mode

**Problem description**:
For DASH pump, the AndroidAPS logo is unexpectedly shown on the DASH tab when on Pod Management (button press)?
Problem is the logo stays in overlay when switching to "Pod History", making history unreadable.

**Additional Info**:

- Logo showing when on  DASH, Pod Management, somewhat unexpected? Could see showing the Omnipod Pump logo here... 
- When selecting "Pod History", logo is showing in overlay affecting readability:

**DASH tab "Pump Managent":**

[image]

**DASH tab "Pump Managent" -> "Pod History", image showing in overlay affects readability:**

[image]

MilosKozak, 2022-12-04:
re-test with https://github.com/nightscout/AndroidAPS/commit/40bbafe14a8aed5a084320421259cb5cad12e636 pls

vanelsberg, 2022-12-04:
@MilosKozak Tested, confirming, fixed

closed 2022-12-05 by MilosKozak