nightscout/AndroidAPS #2251: [DEV][EOPatch] Overlappping temp basals

jbr7rr, 2022-12-04:
Build: 4b46ea8363-2022.12.04 Merged at: https://github.com/nightscout/AndroidAPS/commit/9a35579d66b8f01d0a5610d7485ced848e736f7e

Time: See screenshot (2022-12-04 18:23 and 18:03 would be times of interest)

I noticed basals not cancelling correctly (or not showing up in history at least) causing wrong IOB results. See screenshot:
[image]

Logs:
[AndroidAPS_2022-12-04-jb.zip](https://github.com/nightscout/AndroidAPS/files/10149125/AndroidAPS_2022-12-04-jb.zip)

Will try out some older and newer builds and see what happens

vanelsberg, 2022-12-04:
Can not see the problem here.
E.g. 17:38h 120m basal is replaced with a 5 min basal at 18:03h. Nothing wrong with that...
Please explain?

jbr7rr, 2022-12-05:
> Can not see the problem here. E.g. 17:38h 120m basal is replaced with a 5 min basal at 18:03h. Nothing wrong with that... Please explain?

Please look again, the basal did not end it was overlapping. In fact AAPS was thinking it was still running, the job counter for the one at 18:18 was still going up every 5 min.

Here is another example, please note the end time of the first temp basal. It later then the next one started

[image]

vanelsberg, 2022-12-06:
Any relation with an overlapping problem that was solved for DASH?
https://github.com/nightscout/AndroidAPS/issues/1597

MilosKozak, 2022-12-07:
fixed by https://github.com/nightscout/AndroidAPS/commit/d261bbad09a04d854ee3169eaf927f5502d8c7a2

closed 2022-12-07 by MilosKozak