nightscout/AndroidAPS #2252: profile not working for me in 3.1.0.3 app

sdolenc, 2022-12-04:
I'm having a tough time with profiles in AndroidAPS and NSClient (3.1.0.3).
- I'm unable to download profiles from NS web. ("invalid profile default not accepted from NS" then the "Error" popup is empty)
- I can't upload new profiles from the 3.1.0.3 apps to NS web
- The NSClient app keeps saying "No profile set" even after I create and attempt to activate one.

Any tips for how I can begin investigating/solving this would be greatly appreciated. I'm happy to share information about our setup and I'll use NSClient app to email logs shortly

MilosKozak, 2022-12-04:
PM me your NS address

sdolenc, 2022-12-04:
I just emailed the url to logs@androidaps.org . Let me know if there's a better method of PM'ing you

I'm also comfortable putting this url in this public issue if that's easier

MilosKozak, 2022-12-05:
new valid address is logs@aaps.app but better attach logs here directly
send me your NS address

sdolenc, 2022-12-05:
https://ns.stephener.in/ <-- we plan to eventually change this url so i don't mind it being public for a bit

note: nightscout is dev branch, deployed to a win server 2022 hosted in azure, iis as the reverse proxy, mongo 4.4, node16

logs reproducing the profile issue from nsclient app
[AndroidAPS_LOG_1670180966442.log.zip](https://github.com/nightscout/AndroidAPS/files/10157029/AndroidAPS_LOG_1670180966442.log.zip)

thanks!

sdolenc, 2022-12-06:
Things are getting better! I can now synchronize profile edits from NSClient app to NS web and vice-versa! (not sure if this fixed it: I temporarily pointed NSClient to a new NS web deployment on ubuntu 20.04 and then back to the original NS web)

However, NSClient keeps saying "No profile set" in the app and notifications. This happens regardless of which NS website deployment I use.

Please let me know if there's more info I can provide (NS settings, etc)

sdolenc, 2022-12-06:
Everything's working now!
- per your advice: I attempted profile changes with AAPS full app again instead of NSClient
- I also created a new access token in NS web. I gave it every role I could think of:
- ```"*, admin, api, create, devicestatus-upload"```
- AAPS switched the active profile successfully and now NSClient can do it too.

Thank you for your time and help :+1: