nightscout/AndroidAPS #2257: Too long SMS reply messages

justmara, 2022-12-05:
Currently SMS communicator has 2 major problems with replies:
1. It adds `activePlugin.activePump.shortStatus(true)` to every successful reply
2. It is forced to send successful replies to every number in allowed numbers list.

For some pumps this is not an issue - we had no any pump status in reply messages while were using Combo pump. But for some it is. For example, this is reply message in Russian locale with Omnipod Dash pump:
```
Болюс 0,80ед. подан успешно
Предыдущее соединение: 0 мин. назад
Предыдущии болюс: 0,80 @ 12:04
Временн 0,00U/h @11:41 24/120'
Резервуар: 44ед
```
It contains 142 characters, so it is split onto 3 SMS messages (SMS can contain up to 70 non-ascii characters), and those 3 SMS are sent to each parent. So every successful reply takes 6 SMS messages to be served. Thats expensive :)

MilosKozak, 2022-12-05:
you were not participating when we last time discussed that I'm trying to reduce amount of setting in app to avoid xDrip syndrome (settings for eveything and nobody knows the meaning). both things you describe have good reason
* See status and last bolus time especially when comm fails
* To let know to all caregivers that something already happened

SMS was never meant as a replacement of common care but as a supplement where some SMS doesn't matter
But I understand there can be special cases where it may not fit perfectly

justmara, 2022-12-05:
> you were not participating when we last time discussed that I'm trying to reduce amount of setting in app to avoid xDrip syndrome (settings for eveything and nobody knows the meaning). 

These options are located in particular plugin's settings and have pretty obvious meaning. They help reducing the spam AAPS generates and money drain it (ocassionally) adds. This is much less an issue when using english locale, but translated versions suffer of it much more because of technical limitations of SMS protocol.

> * See status and last bolus time especially when comm fails

Theoretically this can be helpful, but in reality noone cares. Useless bunch of strings that literally costs some money. All that info is in NS and can be seen in nsclient.

> * To let know to all caregivers that something already happened

Sender must receive report, I agree. But the rest can see it in nsclient/treatments. There is no real requirement for this. It is an option not everyone really requires.

> SMS was never meant as a replacement of common care but as a supplement where some SMS doesn't matter But I understand there can be special cases where it may not fit perfectly

SMS **IS** a common care for pumpcontrol mode, for example. Some parents use pumpcontrol as a (literally) remote control for pump, so SMS communicator is a daily option for those.

MilosKozak, 2022-12-05:
on your place i'd change translation to use ascii characters, if SMS are expensive in your country. in europe and most other countries are free or cheap

justmara, 2022-12-05:
> on your place i'd change translation to use ascii characters

You cant only change SMS message's localization - it switches with whole application only. And when people switch app UI language - they do it for a reason, don't you think?

> in europe and most other countries are free or cheap

Some people struggles paying 5$/month for nightscout. SMS messages are (mostly) cheap, but usually comes in packages. I were surprised this month that my kid's aaps has eaten whole 500 sms package only because of switching to Dash (Combo pump were not sending its status for some reason). SMS preboluses are pre pretty common while kid is at school and mother wants to give AAPS some help without setting it overaggressive mode.

It is not a best practice to dictate people what they must to do and how they must do it, instead of simply giving'em an instrument that covers their needs. This code does not change any of current (default) behaviour, but adds an option to customize it to some special (pretty reasonable) needs.

MilosKozak, 2022-12-05:
> It is not a best practice to dictate people what they must to do and how they must do it, instead of simply giving'em an instrument that covers their needs. This code does not change any of current (default) behaviour, but adds an option to customize it to some special (pretty reasonable) needs.
I had similiar opinion whan i started. Now not anymore

MilosKozak, 2022-12-05:
well you are a special case. You try understand the code, improve it etc ......
most people are blind. If you give them option to bypass some safety feature (which you mean "do it only when you know what you are doing" they interpret it as "there is an option and because it's available it's always save to check/uncheck this checbox"