nightscout/AndroidAPS #2262: Reset password is not working

robertrub, 2022-12-06:
I force stop AAPS. I put the PasswordReset file in AAPS\Extra directory. Restart of AAPS does not delete the PasswordReset file and the password cannot be reset.

This was working some time ago.

[image]

gubblebum, 2022-12-07:
Same issue experience here on a live release of AAPS 3.1.0.3, 
Build 1de236a602-2022 10.29-14:46
Flavor: fullrelease
Nightscout version 14.2.6

Steps followed as described by @robertrub including forcestop, PasswordReset file placed in AAPS/extra directoy. No passowrd prompt, and the PasswordReset file remains live in the directory. I tried with a space between the words Password and Reset, as well, to make sure, but to no avail.
[image]

MilosKozak, 2022-12-07:
is the file removed after app restart?

robertrub, 2022-12-07:
No, it is not. It stays in the _extra_ directory.