nightscout/AndroidAPS #2271: Feature request for WearOS: TIR indicator light/color/gauge

DM-LTR, 2022-12-09:
Hi there,

Just an idea - it would be great, if user could use "own color" (in fact half of the display) of watchface "AAPS (Digital style)" as an indicator light for TIR. Just to see he/she is in range within the defined green zone. User-defined colors for hypo/hyper would be welcomed extra.

When doing sports, riding a bike etc. you do not need to know and/or cannot read the exact tiny number values, it is good enough to see "green".

[image]

Or at least some customizable edge gauge similar to Garmins ;) 
[image]

rICTx-T1D, 2022-12-10:
I like your idea.... 

Maybe its a part of solution as bellow:
* create a transparent graphics-file with gray shape on left side
* insert the image  over the background-image as floating layer
* use a function to override the background tint color of gray shape to set a color depending of TIR status
* create a configuration item in menu to activate or disable this function

Andries-Smit, 2022-12-21:
@DM-LTR What about the Simplified UI, together with the always-on?
[image]

When the watch is not active, it shows bigger numbers, the arrow and colour, when activating the watch, it shows the detailed watchface.
https://wiki.aaps.app/en/latest/Configuration/Watchfaces.html#always-on

vanelsberg, 2023-01-03:
Like the idea but do not fancy the proposed color example as it completely alters the watch face looks.
But how about changing the font color for the BG value when BG is out of "range"?

MilosKozak, 2026-10-11:
This can be done with a Custom Watchface (CWF). With dynData, a watchface can change its background colour or images by BG level (low / in range / high). Please see the CWF documentation.

closed 2026-10-11 by MilosKozak