nightscout/AndroidAPS #2272: Pod is not reached!

Viewquest, 2022-12-09:
Hello,
I have the version DEV-D , Build 40bbafe14a since December 4th. installed and get the following error messages:

[AndroidAPS_LOG_1670592620808.log.zip](https://github.com/nightscout/AndroidAPS/files/10195276/AndroidAPS_LOG_1670592620808.log.zip)
I hope the files can be opened and the error can be fixed.

Many greetings
Lothar
[image]

MilosKozak, 2022-12-09:
the crucial log is missing

MilosKozak, 2022-12-09:
that's the same

Viewquest, 2022-12-09:
Ja, welches Protokoll soll ich Dir senden? Habe doch nur das aus der AAPS!

Viewquest, 2022-12-09:
Yes, which protocol should I send you? I only have that from the AAPS!

MilosKozak, 2022-12-09:
it must contain time 14:28 and before

Viewquest, 2022-12-09:
I copied the log right after the error message. May I ask you how I get other log files?

Viewquest, 2022-12-09:
[image]
This error message also comes up. If I immediately take a screenshot, the error should appear. I can't get any other log through AAPS. Or I still have to learn that too. But would like to understand it without a degree in developer language.

MilosKozak, 2022-12-09:
https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html?highlight=accessing%20log

Viewquest, 2022-12-09:
Thank you for this information. So far I have not been able to connect my Android to my Mac. I've only ever sent the data via email. Will now check how I get to the data. Unfortunately I have to pass today. Sorry, I will report back the days when I have solved the problem of data transfer. Many greetings
Lothar

Viewquest, 2022-12-11:
Unfortunately I cannot transfer the logs to the PC. The mobile phone (Pixel 4a) does not allow the transmission. Will clarify that next week. Possibly I'm going to get a new cell phone. Sorry for the inconvenience. Lothar

Viewquest, 2022-12-13:
Since I'm already working with the DEV - E version, it won't make sense to get active here again. With the next issue I will also be able to deliver the corresponding log files. Many thanks for your support.

closed 2022-12-13 by Viewquest