nightscout/AndroidAPS #2274: Failed to import preferences 3.1.0.3 dev D

pagoblan, 2022-12-09:
Hello, I have created a new version (3.1.0.3 dev d), but when I try to import the settings the app closes and no longer restarts, however if I don't try to import the settings, the app works but I lose all the objectives. What I can do? Has it happened to someone?

MilosKozak, 2022-12-10:
you can pm me the export and password.
I received no crash related to this issue

pagoblan, 2022-12-10:
[Archivos Preferencias.zip](https://github.com/nightscout/AndroidAPS/files/10200717/Archivos.Preferencias.zip)
Password 6630
I am attaching the two preferences files that I have used, I have seen that in previous versions, maestro and dev up to C work for me, but in D, the application tries to import them and it closes and does not reopen, leaving a message saying that There is a problem with the application. Phone with Android 13.

MilosKozak, 2022-12-11:
fixed by https://github.com/nightscout/AndroidAPS/commit/1f4d958931a6fa64281d4c701a3bc082cddcc733

closed 2022-12-11 by MilosKozak