nightscout/AndroidAPS #2278: [DEV] Flat data error, when data is not truly flat (xDrip+ with G6)

jbr7rr, 2022-12-11:
Build 91c21b583-2022.12.09 merged at 9c72dced27cb92137011e1beb318c9c481e49254
BGSrouce used: Dexcom G6 with xDrip+ as collector
Time: Flat BG's in log between 05:07:57.078 - 06:27:56.226 (Log's dont go further back, but in screenshots you can tell it was a whole lot longer)
Logs: [AndroidAPS_2022-12-11_jb_flat_data.zip](https://github.com/nightscout/AndroidAPS/files/10201688/AndroidAPS_2022-12-11_jb_flat_data.zip)

The flat bg error get's triggered a lot overnight as you can see in the screenshots. Even when the BG line is not truly flat (there are some small deviations as seen in xDrip+ screenshot). AFAIK this is not the intent of this feature. This can cause a variety in problems, in my case the BG was on target, but this could also happen when BG is not on target and then the BG would stay there because loop is paused and basal is running.

I'm testing the smoothing branch, but this is happening with smoothing turned on and smoothing turned off. And as discussed in the PR the flat BG check is done over the raw value's. (screenshots are taken with smoothing OFF)

[image]
[image]

PS. I use the smoothing branch I know. but this is not directly related to that, so chose to open a new Issue. If it is related please close this and we can discuss it in that PR)

MilosKozak, 2022-12-11:
could you paste screenshot of bg data?

jbr7rr, 2022-12-11:
AAPS bg data (note some data looks doubled, because I switched to BYODA running another sensor, oldest are the xDrip values)
[image]

xDrip+ gives a clearer picture anyway (showing also mg/dl vals)
[image]

swissalpine, 2022-12-13:
Same here:
[image]
[image]

MilosKozak, 2023-01-10:
we should resolve this issue somehow. so the bug in Libre is not repeating anymore and we can remove this test?

jbr7rr, 2023-01-27:
I think removing it completly may leave the door open to problems with the Libre still? Unless we can be sure this is something that will not happen (anymore).

I tried setting the parameters to this:

But that still gave the flat data error too often.

So I see some solutions
1. Remove the check completely which might leave the door open for problems as mentioned above
2. Create something like this: sourceIsDexcom() which checks if it is a dexcom source either from BYODA or xDrip. and exclude dexcom from this check
3. Create something like this: sourceIsLibre() which checks if it's a libre sources and only do the check for that. However this may prove to be quite difficult because the Libre can be sourced in a lot of ways.

For solution 2 or 3 I can create some code and a PR if you like? I think solution 2 would be the best, and could also be used in further checks. I know solution 2 would now be almost equal to isAdvancedFilteringEnabled() but that function has a different intention and might be used differently in the future.

MilosKozak, 2023-02-13:
limit processsing to Libre sources only https://github.com/nightscout/AndroidAPS/commit/34930e323eaea5f3ca9f743c387d071882e9ee6d

closed 2023-02-13 by MilosKozak