nightscout/AndroidAPS #2283: [dev][dynISF] ISF value at 500 if no BG data is available within the past 24h

maheini, 2022-12-12:
## What happened?

I installed AndroidAPS, latest dev (11.12.2022) and somehow my previous values (treatments, BG etc.) got lost during the installation (this isn't the problem). I then enabled DynISF and after the first calculation, DynISF was at 500. it's slightly dropping since then, with a expected (actually credible) value after around 24h of BG values.

### Steps to reproduce

Install AAPS without previous BG values, add new BG source and wait until DynISF is calculated for the first time.

### What I checked

The profile is set, ISF within profile settings is 3.5 mmol. I tried to restart my whole setup, didn't changed a thing.

### System & Logs

Sadly, I don't have a lot of informations (no logs 😢) - but this problem should be easy to replicate.
System: EMUI 12.0.0 -> Android 11
BG unit: mmol

### Screenshot

<details>
  <summary>Open Screenshot</summary>
  
[image]
</details>

DM-LTR, 2022-12-14:
Common behavior after DB reset. Wait and see tha value drops down after a while...

MilosKozak, 2022-12-15:
https://github.com/nightscout/AndroidAPS/pull/2285 waiting for approval

maheini, 2022-12-15:
Thank you Milos 
I just made a new build with this commit and it's working as intended (no ISF set until there is enought data)😄

However, some strange behavior I noticed: After back filling BG & treatments from Nightscout, it's showing me a too low ISF (about half of my profile ISF), even though I was most of the time below target. Might be a problem on my side and I will comment on this if it keeps persistent.

closed 2022-12-15 by maheini