nightscout/AndroidAPS #2287: [dev] Missing BG Value Upload

sfriedriszik, 2022-12-13:
AAPS 3.1.0.3-dev-e
In earlier versions of the Nightscout configuration, the upload of BG values to Nightscout was at the second point.
Now the menu item has disappeared. Rearranged ?

[image]

MilosKozak, 2022-12-15:
it's in BG source preferences. The reason of removal from here was this feature is not available for all bg sources

macarolobe, 2022-12-16:
With fsl2 and xdrip the preference menu is disabled so my bg are not upload to ns

sfriedriszik, 2022-12-17:
I use Juggluco to send the values to Aaps. This is done via the xdrip Broadcase Channel. Now I have no possibility to send the values to Nightscout.

MilosKozak, 2022-12-27:
xDrip is treated as BG source with upload capability thus this option is missing in AAPS

MilosKozak, 2023-01-10:
@sfriedriszik then xdrip can upload the data or?

sfriedriszik, 2023-01-10:
@MilosKozak I don't use xdrip. I use Juggluco with connection to Aaps. Juggluco uses the xdrip broadcast to transfer the data to Aaps.

MilosKozak, 2023-01-10:
if you install xdrip it will receive the data as well

sfriedriszik, 2023-01-10:
@MilosKozak  I know that. I don't want to install xdrip. that worked great before. the diabox users have the same problem.

MilosKozak, 2023-01-10:
you may ask juggloco devs to create a special broadcast to aaps

closed 2023-01-21 by MilosKozak