nightscout/AndroidAPS #2288: [3.1.0.3] Dismissing Android Keyboard Closes Wizard Before Completion

old-square-eyes, 2022-12-13:
On my S22 Ultra, after filling out the bolus wizard, the Android keyboard (SwiftKey) sits over the OK button on the confirmation screen.

Dismissing the keyboard closes the wizard before you can hit OK.

Also happens on carb/insulin entry screens.

[image]

MilosKozak, 2022-12-15:
this is purely android thing. how could i help?

old-square-eyes, 2022-12-15:
@MilosKozak you say like I'm supposed to know already it's an Android thing. It never used to do this. It seems like a bad UI bug. Maybe it's Android, sure. But I don't know that, so I reported it. Besides, whether Android or not there may be a way round it. Sounds like you're saying there isn't.

old-square-eyes, 2022-12-15:
Besides that. There are things you can do as an app developer to work around it. Maybe it's not a priority for you, which is fine. But it abruptly got worse for me. Maybe it's the S22 or maybe Android update. But it's been an issue for one a couple of months now. I guess if I reported it others could also add thier experience at the very least.

Andries-Smit, 2022-12-17:
@old-square-eyes it seems keyboard related, when you switch to default Samsung keyboard, it does not happen. Can you confirm? 
I'll check why it happens.

old-square-eyes, 2022-12-17:
Just tried with Samsung keyboard and it does the same.

Andries-Smit, 2022-12-19:
@old-square-eyes This issue seems to be an Android bug. I tested it and could reproduce the issue. Now I updated to the latest version (patch 1 Dec 2022) and it does not seem to occur anymore. Could you check if your phone is up to date?

old-square-eyes, 2022-12-19:
I'm on 1 Nov (no updates pending). Guess I'll have to wait until Samsung
pushes mine to re-test. Thanks for looking at it though. Much appreciated.

old-square-eyes, 2022-12-26:
Just got OTA and it's fixed for me. Thanks for the help @Andries-Smit

closed 2022-12-26 by MilosKozak

old-square-eyes, 2023-01-19:
This has been happening again. I'm not sure why TBH. Maybe my test was flawed. I'll post a video.