nightscout/AndroidAPS #2289: Feature request: Menu setting "Enable simple bolus calculator"

amadangel, 2022-12-13:
**TL;DR** Add an optional setting "Enable simple bolus calculator" to have the bolus calculator open without any ticks (BG, Trend, IOB, COB) set by default

**Problem description:**

Many people have day-to-day varying insulin needs (e.g. growth periods, hormonal, sickness) that are not well captured by current sensitivity detection. One way out is to adapt the profile manually as soon as a pattern appears. This is often not possible, in particular for children with less experienced caregivers during most of the day.

One consequence of a badly matched profile is an accumulation of IOB (positive in the case of insulin resistance or negative in the case of increased sensitivity). When using the bolus calculator in such a situation, the wrongly accumulated IOB/COB makes matters worse when bolusing more insulin is given at times of sensitivity and less is given at times of resistance. A practical solution is to manually remove all ticks of the bolus calculator and only consider carb input for bolusing.

While experienced users can do this easily, we have found it impossible to teach this to our 8 year old or her caregivers. This has repeatedly caused hypos or hypers. To make things worse - after removing ticks once, the bolus calculator currently defaults to opening with only an IOB tick that is not matched with a BG and COB tick. In the case of resistance, this makes the bolus suggestion too small by an even bigger margin and contributes to lasting highs.

**Proposed solution:**

A “simple bolus calculator” option in the bolus calculator menu settings that makes the bolus calculator open with all ticks OFF by default. This, in many situations would be a better and safer option than the current default behavior, at least for children in an out-of-home-setting and others with frequently varying insulin needs who rely on the bolus calculator.

**Proposed implementation:**

Menu:
In the menu Preferences > Overview > Advanced settings add the following toggle switch:

> Enable simple bolus calculator
> Use this setting to open the bolus calculator with all ticks OFF by default. Default bolus suggestions will only be based on carb input. BG, trend, IOB and COB information can be added into the bolus suggestion manually.

Behavior:
When setting is off (default): ticks in calculator have current behavior
When setting is checked: calculator opens with no ticks checked. Any ticks placed will not be remembered for the next use of the bolus calculator

**Additional comment:**

I cannot help with the implementation in the code, but I would be happy to contribute in any other way. It’s a feature that would make our life easier and having our kid in school considerably safer, and enable bigger independence from the “master caregiver”. Thank you for considering!

Magdalena ( discord Magdalena#1674 , using dev & dynISF + so grateful it exists!)

amadangel, 2022-12-17:
Today I installed AAPS wear and saw that the above is a feature of the watch version bolus wizard - which is perfect!!
in some sense I hope this will solve it for us - but why not move it to the main app?

juehv, 2023-11-07:
I have a similar wish, but with some more visuals to explain the proposed bolus. I'd implement it as an alternertive Bolus-Wizard.
I can do the work *if* it has any chance to be merged and nobody else already started :)

amadangel, 2023-11-11:
Depends on the situation and settings. In some situations (low bg) or depending on SMB settings, SMB will not do this, but the bolus calculator would still do it.

juehv, 2023-11-13:
Exactly. This is one of the "Loop tries to kill me" situations, were the bolus calculator suggests pushing more insulin in a hypo ...
[image]