nightscout/AndroidAPS #2290: Automations : change max profile percentage hard limit

robertrub, 2022-12-14:
For some meals, mainly cheesy (as raclette, fondue, tartiflette... I live in a cheese production region ;). ) the BGs raise high late after the meal because of all that fat and protein.

I have an automation that raises profile % for 10 min to help bring down the BGs.  The basic profile is not strong enough for the job (FIASP is less efficient over 200+ too).

The hard coded maximum 130% is not sufficient. At least 150% or, even better 180% limit would be much better. Actually, I have to set it manually but at night, it is not easy.

I can make a stronger copy of my profile and switch to that one. But, as I tweak my profile often, I'd need to change both profiles with risk of errors.

If 130% is set for a security reason, it would be more efficient if the security was on the duration of the action and not the percentage.

MilosKozak, 2022-12-15:
This is not what automation was designed for

robertrub, 2022-12-15:
I use it for exercise with 50% profile but would be helpful to use it for a high fat/protein meal too ;) Actually, I'm setting it manually (if I realise rapidly that BG are going too high). If I don't realise it, the BGs go over 200+ and then FIASP is les effective and stuck with hours of high BGs :(

Andries-Smit, 2022-12-17:
For high fat protine meals Extended Carbs should be used. 
Source of the Fat Protine Unit (FPU) calculation research https://www.ncbi.nlm.nih.gov/pmc/articles/PMC2901033/

robertrub, 2022-12-17:
@Andries-Smit  The extended carbs just add carbs but does not inject the required bolus that goes with them :( The SMBs try to play catch up but is not enough if the basal rate is not raised.

closed 2022-12-25 by robertrub