nightscout/AndroidAPS #2291: [3.1.0.3-dev-e] - Master and NSclient Apps shows persistently 50U of remaining insulin...

streethawk22, 2022-12-14:
NSclient App and Master App too, shows 50U of remaining insulin in reservoir. On Screenshot see you it better.

THX...
[image]

--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

streethawk22, 2022-12-14:
Ahoj Miloš,Neviem, či to suvisi s pumpou ako takou, ale pumpa je to Dana-i...Peter HolubekOdoslané z môjho zariadenia Galaxy S22 Ultra.
-------- Pôvodná správa --------Od: Milos Kozak ***@***.***> Dátum: 14. 12. 2022  13:30  (GMT+01:00) Komu: nightscout/AndroidAPS ***@***.***> Kópia: streethawk22 ***@***.***>, Author ***@***.***> Predmet: Re: [nightscout/AndroidAPS] [3.1.0.3-dev-e] - Master and NSclient Apps shows persistently 50U of remaining insulin... (Issue #2291) 
What pump?

MilosKozak, 2022-12-15:
fixed by https://github.com/nightscout/AndroidAPS/commit/31a6bf446033be4d17c14864843403618772417d

closed 2022-12-15 by MilosKozak