nightscout/AndroidAPS #2293: NSClient is not syncing to database whilst xdrip is

mcurly, 2022-12-17:
After pulling up the "stunt" of ("almost") having my database inside my own premises (inside a docker container managed within portainer), there seems to be something preventing AAPS NSClient from syncing into that Docker container instance.

(Meanwhile) xDrip is able to sync without issues.

MilosKozak, 2022-12-17:
based on the info you provided what kind of help are you expect?

mcurly, 2022-12-17:
None deserved right? None will be provided I guess.

Sorry for the inconvenience @MilosKozak . That's "fatigue" all over myself.

Well, I will try to replicate the steps of deploying docker images again. (Inside portainer and with an adjusted docker compose file for Nginx Proxy Manager, without the original traefik load balancer that you provide here in the repos).

Since I removed the containers, by now, I only have the logs from xDrip and AAPS, which aren't 'anonymized'.  Do you 'expect' me to share them with you? What shoud be the channel for that disclosure? I can use of means to anonymize them to an extent. Can you please show some sympathy and enlighten myself/point to procedures to how should I share that info?

Best regards dear @MilosKozak and I apologize to not having shared all the convenient information so far. Blame myself and my tired frustration sprung out of a late night marathon to make this setup of mine work.

P.s.: I hope I can somehow write a roadmap/document of the situation, after having a minimum set up accomplished (meaning: working as expected).

Thank you.

mcurly, 2022-12-17:
Hi again @MilosKozak ,

I hope you will be somehow pleased to know that my case can be closed. It is working now as expected.

I apologize for being a nuisance of a fellow.

Again, I hope I can write out a guide somewhere (perhaps in the documentation) to how did I adapt your default traefik setup to my Nginx Proxy Manager one.

Two major steps that I am positive I did diferently this time:

1. In the first attempt I only dumped nightscout database from the mongodb cluster laying inside the googlecloud infrastructure. I don't know if this would be relevant to this, but since I did this step differently this time, because I exported all of the databases from googlecloud and restored them to my docker container one.
Perhaps the fact that the 'spelling' of the main datase name is different ( '**N**ighscout' in google cloud VS '**n**ightscout' inside docker compose yaml file), had something to do with this (and this is only a speculation), the other one being that in this second attempt I 'dumped' (from gcloud) and 'restored' (into docker), **all** the databases from gcloud VPS instance.

2. The second different thing that I suspect was the responsible for this more welcoming status of my docker deployment stack, was the fact that _I triggered the 'web-sockets' switch to 'on'_ inside the configuration panel for this 'proxied' docker container.

Now I am going to close this topic.

And again, thank you @MilosKozak  and everyone that contributes to this 'flabbergasting' project!!!

Cheers!

closed 2022-12-17 by mcurly