nightscout/AndroidAPS #2294: Feature Request: Automation Profile Switch by day without specifying time

happygirlnyc70, 2022-12-17:
Right now there is no way to set up an automation to do a profile switch that differentiates for just days without specifying a time if you want to choose between more than one profile.  For example: I have a profile for the work week and and one for the weekend. Those are set up to switch at a specific day and time. I also I have profiles for when I'm sick or have some type of hormonal changes. Right now I have to choose my work week profile because I can't choose just days without also picking a time. So if I'm sick starting Friday but start to feel better on Sunday right now my profile will change from my sick profile to my work week profile because I can't say Monday through Friday use work week but Saturday and Sunday use weekend.

Thanks for the consideration and all of the hard work that goes into making my life and so many other lives so much better!

Andries-Smit, 2022-12-21:
@happygirlnyc70 Can you explain a bit more about what you need? 
Do you need a condition trigger type in the automation with "day of the week"? Just like "Time Range" selecting days of the week without a time?
Or do you need to prevent the Time Range selector from triggering when you are on your "sick" profile? In that case, you might need a trigger "profile" not "sick"....

happygirlnyc70, 2022-12-21:
@Andries-Smit, Your first option is what I am asking for. I need an automation condition trigger for "day of the week" without time.

I have four profiles. Two profiles that change at regular time intervals based on the day of the week; Daily and Weekend basal profiles. Daily and Weekend are my normal profiles. I also have two profiles that change based on insulin resistance/sensitivity; Sensitive and Resistant basal profiles. Sensitive and Resistant can be triggered any day of the week and at any time (I used being sick as an example). Right now I don't have a way to revert back to either Daily or Weekend based on just the day. I have to choose for it to always revert to daily profile or weekend profile because the day of the week condition also requires a time trigger.