nightscout/AndroidAPS #2296: Suggestion: Add optional alert for "Low phone battery"

1401Ger, 2022-12-18:
Since many phone operating systems don't provide a setting for a customizable low battery alert (e.g. battery <10 %) I think it would be very helpful to have this as an option in AAPS.
To give an example, if the phone is shutting down during the night due to low battery,  the user will not get any alerts for low or high BS levels and obviously there is no looping anymore. While the pump will revert to it's basal profile, the chance of ending up in a hyper for several hours without the user noticing is far from ideal. Such a scenario could easily be avoided by a simple low battery alert.

maheini, 2022-12-18:
You are able to make this warning with apps like tasker or automate. I don't know if it really is a thing which AndroidAPS should be responsible for - I'd rather say no and keep AAPS "simple".

1401Ger, 2022-12-18:
> 
I know there are workarounds for it, I just think that AndroidAPS is a tool that a user has special requirements for in terms of reliabilty und robustness. Adding this option for an alert is not cluddering the app in any way (would be in the existing alerts/warning menu) and is probably easy to integrate.

swissalpine, 2022-12-20:
My smartphone does that anyway, it doesn't need to do AndroidAPS ...

1401Ger, 2022-12-20:
`My smartphone does that anyway, it doesn't need to do AndroidAPS ...`
Glad to hear that your smartphone provides that functionality already. However Android 12 does not.

Andries-Smit, 2022-12-21:
@1401Ger what phone are you using?

1401Ger, 2022-12-21:
@Andries-Smit Samsung Galaxy S10e. And just to clarify: The issue isn't that this (and many other smartphones) don't give the user a notification when the battery is getting low, but there seems to be no way to customize it or make it an audible alert that would wake you up during the night for example.

MilosKozak, 2022-12-24:
AAPS should be a loop app. Even I understand you may be missing it, this is not feature that should be provided by AAPS itself

closed 2022-12-24 by MilosKozak