nightscout/AndroidAPS #2298: Yellow triangel - no data?

TinoLL, 2022-12-21:
Using 3.3.0.3 and Android 13 and FSL2. Sometines the main screen shows "no data" for 521 minutes, 412, ... (always different times). Going into the bolus calculator the newest value coming from XDrip is in. 
Not 100% certain but this occures when the phone is in flight mode (BT is still on). XDrip receives data. 

And there is always a yellow triangle at the screen. Clicking on it and doing re-calculation doesn't help. 
[image]
[image]
[image]

maheini, 2022-12-23:
I'm not experiencing similar behavior (3.1.0.3, guess you meant this version) - but I haven't tested with flight mode enabled.

However, the yellow triangle is "normal" for libre 2, since the readings from xDrip are not sent exactly every 5 minutes.

What happens if you re-run the loop manually? And can you provide a screenshot of the xDrip readings within AAPS while experiencing this issue? (Enable the xDrip tab in Config Builder and open this tab)

TinoLL, 2022-12-24:
Sorry, yes: 3.1.0.3. 
OK, the yellow triangle is caused by the Libre 2. I understand, Thanks for explaining. And it seems to be independent of the flight mode because it happended during daytime (without flight mode) as well. 

As usual it didn't appear anymore. Next time I will do a screen shot from XDrip. And data returns after a while. Why ever.

Another thing yesterday (same cause????): I scanned with the L2 patched app. This raw value was set to AAPS although I use calibrated values. With the next standard reading over BlueTooth the calibrated value was used.

closed 2022-12-24 by MilosKozak

TinoLL, 2022-12-25:
@MilosKozak What is the solution for the missing data, please?

MilosKozak, 2022-12-26:
I don't know. you may test dev branch (if you know how to do it). There are many bugs of this kind fixed

TinoLL, 2022-12-29:
I found Def-F as newest. May I use this one? Will it solve my issues?