nightscout/AndroidAPS #2299: Dev-e: Omnipod Dash: AAPSCLient showing conflicting remaining amounts

kristinsmotz, 2022-12-22:
In new AAPSclient after installing dev-e (possibly also dev-d), there are two displays of remaining amount in Dash pod on the overview tab. The top one constantly displays 50U even when remaining amount is less. Correct remaining amount corresponding with AAPS phone Dash tab displays in next line of icons.
[image]

[image]

MilosKozak, 2022-12-23:
this is already fixed i recent devs

kristinsmotz, 2022-12-23:
Perfect :)

kristinsmotz, 2022-12-30:
I noticed that the newer version now displays remaining amount only on top line, not in the Pump section of the lower line. And when remaining amount goes to 0U, it disappears completely from both lines. It also disappears in AAPS app. 
Have updated to dev-f today and it's the same there. No number at all for 0U. 

AAPSClient: 
[image]

AAPS:
[image]
[image]

kristinsmotz, 2023-01-03:
I noticed that the newer version (dev-f) now displays remaining amount only on top line, not in the Pump section of the lower line. And when remaining amount goes to 0U, it disappears completely from both lines. It also disappears in AAPS app. 
Have updated to dev-f today and it's the same there. No number at all for 0U. In the lower line, UPLD (uploader phone battery%) disappears for me when it reaches 100%, maybe there is too much padding in that section/careportal info line?

AAPSClient: 
[image]

AAPS:
[image]
[image]

MilosKozak, 2023-01-10:
is this working well now with latest dev?

kristinsmotz, 2023-01-10:
Unfortunately there is still a discrepancy. I just rebuilt from dev now after your update and this is a screenshot from AAPSClient. 
[image]

As you can see, the bottom line shows 0.0U and some superfluous text, while top line correctly shows more than 50U. I did wait until next glucose/loop run after installing the update to see, but still 0.0U.

MilosKozak, 2023-01-10:
https://github.com/nightscout/AndroidAPS/commit/f32709eaa6c30b64681c1e60a8e46db293db0f18

closed 2023-01-10 by MilosKozak