nightscout/AndroidAPS #2300: [DEV]

jbr7rr, 2022-12-23:
Build 5d9155daab-2022.12.23 merged at 0f5d3b5813b591613c21b0ec5dbd55c2498a0313
Time: 2022-12-23 21:185
Logs:  [AndroidAPS_2022-12-23_JB.zip](https://github.com/nightscout/AndroidAPS/files/10297325/AndroidAPS_2022-12-23_JB.zip)

For some reason AAPS keeps syncing (resending) my profile with NS 2 times per minute, resulting in my database growing larger and larger...

I use the smoothing branch, but another user reported the same problem on discord using dev:
https://discord.com/channels/629952586895851530/629954570394533889/1055885334149349377

[image]

MilosKozak, 2022-12-23:
maybe this https://github.com/nightscout/AndroidAPS/commit/dbc73f22f42988c9623128e66a94b015962fc9ae fixed it?

jbr7rr, 2022-12-24:
It did not unfortunately.

reverting back to old dev (2022.12.01) and back fixed it until I change one of my profiles.. But that also means I can reliably reproduce this for myself

Alamo04, 2022-12-24:
Confirmed.
The last working version is from 09.12.2022 00:02. I stay now with that version.

MilosKozak, 2022-12-24:
profile upload is not acked by NS.
full DB?
it's working well with my setup

mick711, 2022-12-24:
I test the dbc73f2 because I have a problem like that but it's possible my database the problem because I use nightscout on my nas synology. All data was resent... 
[image]
[image]

Alamo04, 2022-12-24:
If you have not mutch space in your NS Database, you will become a problem, there are thousands of profile entry's after that issue comes up. See picture, that are only a few of the entrys and there are so mutch, that you can't delete them with NS tools.

[image]

Alamo04, 2022-12-24:
And @Milos
- If you don't change the profile, everything works fine.
- If you change the profile, spamming starts.

I'm now on dev from yesterday, but must first install and start an old dev (09.12.2020). After that I can go back to the actual dev until I don't change something in my profile.

jbr7rr, 2022-12-24:
I did clean my NS this morning and removed all doubled profiles, there is enough space now. I changed some profile around 14:00 and it started spamming again.

If it is not acked it might be a problem on the Nightscout side? I can open a ticket there if you like..

New logs for good measure:
[AndroidAPS_LOG_1671886923881.log.zip](https://github.com/nightscout/AndroidAPS/files/10298636/AndroidAPS_LOG_1671886923881.log.zip)

MilosKozak, 2022-12-26:
how much profiles do you have in AAPS?

Alamo04, 2022-12-26:
3 at me.

MilosKozak, 2022-12-26:
pm me your NS and secret. I'll try to reproduce

Alamo04, 2022-12-26:
Sendet over discord.

jbr7rr, 2022-12-26:
Sent over Discord as well

Alamo04, 2022-12-27:
@MilosKozak 

Just now tested the last Dev version:
With that version you didn't need a profile switch to force this issue, this version starts spamming  directly after upgrade.

It takes only 2 minutes, before I was back to the old version, but in that time it is spamming so fast, that you can't delete the entries manually.  

Have to delete all Profiles over 10be again.

My System:
- Google Pixel 5
- Dexcom G6 over Boyda
- BG data over AAPS to NS
- XDrip only for the Smartwatch  (no uploads)
- No downloads from NS are allowed in AAPS

jbr7rr, 2022-12-27:
For all other users that are affected by this bug, I have created a small python script that removes the duplicates from your mongo db. Use at your own discretion and use only if you know what it does, as this will edit your mongo db. It also can take quite a while to run when there are lots of duplicates
```
from pymongo import MongoClient
import pprint

# Replace the uri string with your MongoDB deployment's connection string.
uri = "mongodb+srv://<user>:<password>@<cluster-url>?retryWrites=true&writeConcern=majority"

client = MongoClient(uri)

# database and collection code goes here

db = client.ns0 # REPLACE ns0 with your database name!
collection = db.profile

prevStartDate = ''
for doc in collection.find():
    #pprint.pprint(coll.find_one(doc)) 
    profile = collection.find_one(doc)
    
    startDate = profile['startDate']
    if startDate == prevStartDate:
        print('duplicate found, removed!') 
        print('startDate: ' + startDate)
        collection.delete_one(doc)
    # We delete one by one so one original copy is still left..
    prevStartDate = profile['startDate']

# Close the connection to MongoDB when you're done.
client.close()
```

MilosKozak, 2022-12-27:
> @MilosKozak
> 
> Just now tested the last Dev version: With that version you didn't need a profile switch to force this issue, this version starts spamming directly after upgrade.
> 
> It takes only 2 minutes, before I was back to the old version, but in that time it is spamming so fast, that you can't delete the entries manually.
> 
> Have to clear the Profiles Database over 10be again.
> 
> My System:
> 
> * Google Pixel 5
> * Dexcom G6 over Boyda
> * BG data over AAPS to NS
> * XDrip only for the Smartwatch  (no uploads)
> * No downloads from NS are allowed in AAPS
> 
> **Edit 27.12.2022 13:50** Next try to fix it.
> 
> Build a complete new Dev and did not only update it over Git.
> 
> * now the old issue is back, but spamming starts first after profile switch again and only all 30 seconds.
> * So I can use this version, until I need a profile switch.

resolved by https://github.com/nightscout/AndroidAPS/commit/43cef1e7b631c6efcaf4f9dbe1f8ff59c7d6044b
but this is not related to multiple uploads od profile store

Alamo04, 2022-12-27:
Not resolved.

Still uploading profiles every queue, after profile switch.

MilosKozak, 2022-12-27:
profile store upload fixed by https://github.com/nightscout/AndroidAPS/commit/3472f0f83e81164869453e580094677ceb1e5b73

jbr7rr, 2022-12-27:
Can confirm, for me it works now as it should! Many thanks!

closed 2022-12-27 by MilosKozak

Alamo04, 2022-12-27:
Confirmed, works now. 👌