nightscout/AndroidAPS #2302: time of sync on xDrip and AAPS 3.1.0.3-dev-f dont match

idalioreis, 2022-12-26:
last AAPS verson DEV  (3.1.0.3-dev-f), the time between sensor reading is not sync on xDrip and AAPS even with Smoothing disabled. AAPS reading from xdrip

Xdrip 09.12.2022

[image]
[image]

ct2iqj, 2022-12-26:
I have de same problem.

MilosKozak, 2022-12-26:
not possible to resolve without logs
what cgm?

MilosKozak, 2022-12-27:
... and timestamps on xDrip tab match xdrip?

idalioreis, 2022-12-27:
> not possible to resolve without logs what cgm?

CGM is libre2

with smoothing on, the time of reading on aaps increase
[image]
[image]
[AndroidAPS_LOG_1672132132250.log.zip](https://github.com/nightscout/AndroidAPS/files/10307298/AndroidAPS_LOG_1672132132250.log.zip)

with smoothing off and when xdrip read sensor (0m) the AAPS change to 1m
[image]
[image]
[AndroidAPS_LOG_1672133977692.log.zip](https://github.com/nightscout/AndroidAPS/files/10307301/AndroidAPS_LOG_1672133977692.log.zip)

[image]
[image]

MilosKozak, 2022-12-27:
there is BG every 6 min in AAPS

MilosKozak, 2022-12-27:
I checked log
```
06:57:22.068 [androidx.work-4] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=1.7834293414914097E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=116.48398911466458, com.eveningoutpost.dexdrip.Extras.Time=1672124241706, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672041037019, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=116.4839891146646, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
07:03:22.802 [androidx.work-3] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=1.7820248795108606E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=117.12664431101236, com.eveningoutpost.dexdrip.Extras.Time=1672124602338, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672041037019, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=117.12664431101236, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
07:09:23.054 [androidx.work-2] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=2.0068313867680362E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=117.84963140690357, com.eveningoutpost.dexdrip.Extras.Time=1672124962601, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672041037019, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=117.84963140690357, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
07:15:23.503 [androidx.work-1] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=1.5601326648899517E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=118.41195470370785, com.eveningoutpost.dexdrip.Extras.Time=1672125323034, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672041037019, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=118.41195470370785, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
07:21:24.217 [androidx.work-4] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=-3.789040789300713E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=117.04631241146886, com.eveningoutpost.dexdrip.Extras.Time=1672125683453, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672041037019, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=117.04631241146888, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
07:27:24.230 [androidx.work-2] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=4.4580413796944785E-7, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=117.20697621055582, com.eveningoutpost.dexdrip.Extras.Time=1672126043844, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672041037019, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=117.20697621055582, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
07:33:25.082 [androidx.work-4] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=4.963261193063715E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=115.76501900612938, com.eveningoutpost.dexdrip.Extras.Time=1672126404719, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672126133446, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extr [... cut ...] drip.Extras.Time=1672133971314, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672126133446, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=FortyFiveDown, com.eveningoutpost.dexdrip.Extras.BgEstimate=177.4579785181022, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
```

xDrip is sending data every 6 mit which is not supported by AAPS

idalioreis, 2022-12-27:
> I checked log
> 
> ```
> 06:57:22.068 [androidx.work-4] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=1.7834293414914097E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=116.48398911466458, com.eveningoutpost.dexdrip.Extras.Time=1672124241706, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672041037019, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=116.4839891146646, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
> 07:03:22.802 [androidx.work-3] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=1.7820248795108606E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=117.12664431101236, com.eveningoutpost.dexdrip.Extras.Time=1672124602338, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672041037019, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=117.12664431101236, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
> 07:09:23.054 [androidx.work-2] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=2.0068313867680362E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=117.84963140690357, com.eveningoutpost.dexdrip.Extras.Time=1672124962601, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672041037019, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=117.84963140690357, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
> 07:15:23.503 [androidx.work-1] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=1.5601326648899517E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=118.41195470370785, com.eveningoutpost.dexdrip.Extras.Time=1672125323034, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672041037019, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=118.41195470370785, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
> 07:21:24.217 [androidx.work-4] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=-3.789040789300713E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=117.04631241146886, com.eveningoutpost.dexdrip.Extras.Time=1672125683453, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672041037019, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=117.04631241146888, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
> 07:27:24.230 [androidx.work-2] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=4.4580413796944785E-7, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=117.20697621055582, com.eveningoutpost.dexdrip.Extras.Time=1672126043844, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672041037019, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=117.20697621055582, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
> 07:33:25.082 [androidx.work-4] D/BGSOURCE: [XdripPlugin$XdripWorker.doWorkAndLog():101]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=4.963261193063715E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122122416 689d22a-2022.12.24, com.eveningoutpost.dexdrip.Extras.Raw=115.76501900612938, com.eveningoutpost.dexdrip.Extras.Time=1672126404719, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1672126133446, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.evenin [... cut ...] iseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=FortyFiveDown, com.eveningoutpost.dexdrip.Extras.BgEstimate=177.4579785181022, com.eveningoutpost.dexdrip.Extras.SourceDesc=LimiTTer}]
> ```
> 
> xDrip is sending data every 6 min which is not supported by AAPS

Only dev-f start to do this, the previous version work well.
Can you see the commit from @jbr7rr, if ir is One solution.
Thanks

mjhuisman, 2022-12-27:
Got the same since updated dev to latest commit.
Xdrip is sending bg every 5 minutes, but bg of aaps is different from bg of xdrip.
I use fsl2 with patched app.

macarolobe, 2022-12-27:
Same problem with smooth or without[image][image]

mjhuisman, 2022-12-27:
Reinstalled previous version from 5 days ago and everything fine.

MilosKozak, 2022-12-27:
the mode has changed. 
Before smoothing AAPS displayed BG like it was received from BG source
Since smoothing is in place AAPS displays bucked data ie data after recalculation to 5 min like they are used in algorihtm.
Isn't it the reason?
Anyway this issue has a reason of 6 min data.
If it's not your case create another issue with screenshots, logs atc

mjhuisman, 2022-12-28:
It has 6 min in my case also, but the main problem is that between 3 bg readings aaps isnt working because of no bg and/or no delta s.

MilosKozak, 2022-12-28:
resolve primary issue first. you need 5 min or less

macarolobe, 2022-12-28:
I'm 5 minutes (and one or two seconds) data and even with no smoothing there is a delay beetween xdrip bg and aaps bg. It's not always, sometimes there is no delay when bg is flat. It seems that the delay increase when bg increase or decrease.

mjhuisman, 2022-12-28:
> resolve primary issue first. you need 5 min or less

How?

MilosKozak, 2022-12-28:
go to xDrip groups

MilosKozak, 2022-12-28:
> I'm 5 minutes (and one or two seconds) data and even with no smoothing there is a delay beetween xdrip bg and aaps bg. It's not always, sometimes there is no delay when bg is flat. It seems that the delay increase when bg increase or decrease.

then create another issue with screenshots, logs etc

parapenT1sta, 2022-12-28:
> > resolve primary issue first. you need 5 min or less
> 
> How?

Are you using OOP2? Maybe we need to ask xDrip developers to check if there is any issue due OOP2.

idalioreis, 2022-12-28:
> > > resolve primary issue first. you need 5 min or less
> > 
> > 
> > How?
> 
> Are you using OOP2? Maybe we need to ask xDrip developers to check if there is any issue due OOP2.

OOP2 read sensor evrey 2m, so 3 read give 6m, maybe OOP2 or bluetooth issue

[image]

idalioreis, 2022-12-28:
Until now its working fine, the issue was fixed at newest version. thanks

closed 2022-12-28 by idalioreis