nightscout/AndroidAPS #2304: TempTarget-triggered not respected in loop run

justmara, 2022-12-27:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-12-27:
what version is this?

justmara, 2022-12-27:
Im running master for my kid.
You say this behaviour changed in current dev? Will wait for next release then, being careful. It was first time I'v met such behaviour and it is a really weird one.

MilosKozak, 2022-12-27:
👍 sequetial procesing caused more issues not only this one. But they were random and not easy to catch

closed 2022-12-27 by MilosKozak

justmara, 2022-12-27:
> Here is this problem that loop was triggered by 1st change before second was applied

Doesn't seem to be. Loop run has read for treatments and did not found both these TT records for some reason. Even though hypo TT was set long time ago. I've checked treatments log in AAPS and it shows original TT set time, not the real one applied by AAPS - that could be the problem too. While nsclient and NS see the TT, AAPS can be missing it, but when it receives - it adds it to log like it were there all that time.