nightscout/AndroidAPS #2317: xDrip BG not up to date in AAPS

robertrub, 2022-12-31:
I think that this was signalled before but couldn't find the issue.

BGs coming from xDrip. The BGs are correctly shown in the xDrip tab in AAPS but the main UI shows older BG.

[image]

[image]

jbr7rr, 2022-12-31:
Which build? Logs?
A similar issue was fixed a few days ago

robertrub, 2022-12-31:
[image]

MilosKozak, 2022-12-31:
scroll down on BG tab, there is probably irregularity in last 2 days

robertrub, 2022-12-31:
@MilosKozak I changed transmitters

jbr7rr, 2022-12-31:
Yes after transmitter change, the data is not in 5 minute interval and gets recalculated, that causes the delay. However after your build, changes has been made to make this delay less frequent, so try to update to latest Dev and see if that fixes it

robertrub, 2022-12-31:
I've just built the last dev now. But the problem happens ONLY if using xDrip. Using BYODA, everything is in sync.

I'll see in 10 days for the next sensor change (I'll switch to xDrip and 2nd transmitter)

idalioreis, 2023-01-01:
what i detect is the trend at xdrip are not the same at AAPS, so i cant guide by AAPS trend, i have to go always  to xdrip.

robertrub, 2023-01-01:
@idalioreis The different trends in xDrip and AAPS is "normal" as xDrip shows the arrow based on the last 2 BGs while AAPS uses a longer delta (or the one produced by Dexcom, can't recall exactly)

MilosKozak, 2023-01-10:
https://github.com/nightscout/AndroidAPS/commit/4b6eab6aa405b6906d7eb15fa1458759ab9dc3df

closed 2023-01-10 by MilosKozak