nightscout/AndroidAPS #2319: [Latest DEV] Request: Backfill BG values with "grey" color?

parapenT1sta, 2023-01-01:
Today replacing the sensor, I realized that after the two warm up hours a straight line was created from the last BG value before the sensor removed to the new value after two hours. Is it possible to put these values in grey or other color? Since they aren't real BG readings?

Using G5 with BYODA. The readings aren't sent to xDrip as screenshot below.

Screenshot of the straight line in AAPS:
[image]

in Xdrip:
[image]

kristinsmotz, 2023-01-05:
Seen this for the first time today as well with our G6 sensor change. Love that values are filled via smoothing where there would otherwise be no data, but I agree especially when filling larger gaps they could be a different color to clarify that they're not "real data".

parapenT1sta, 2023-01-21:
@MilosKozak This was already done. correct?

https://github.com/nightscout/AndroidAPS/commit/556c1a048fa591a28afebd7ad3752f98be4ac09f

MilosKozak, 2023-01-21:
yes

closed 2023-01-21 by MilosKozak