nightscout/AndroidAPS #2323: [3.1.0.3] Wear entered treatments are not received by my phone (no confirmation)

old-square-eyes, 2023-01-02:
Wear entered treatments are not received by my phone (no confirmation)

Keys should be the same as was working fine for a month.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem. 11:05am NZDT
- Logs Sent
- FB Post: https://www.facebook.com/groups/AndroidAPSUsers/posts/3438350919719634

vanelsberg, 2023-01-03:
Important information is missing on this issue:

* No logs included with this issue
* What AAPS release/version/build are you running (add a screen print from AAPS's about?
* What phone brand/type/OS
* What watch brand/type/OS
* Are Phone and Watch running latest updates
* How did you install the wear-full-release.apk?
* What watch face are you using?

Did you try uninstalling the wear app and select a stock watch face, then switch off the watch, then switch it on again and reinstall the AAPS wear app?

old-square-eyes, 2023-01-04:
@vanelsberg

- logs were shared via email from the app. I also referenced this Github issue number in the email. See "Logs sent" above. Sorry if I did it wrong.
- Version number is in the title.
- Phone S22 Ultra, Galaxy Watch 4 (both on latest updates)
- Wear app installed via "Wear Installer 2". Was working for treatments then unexpectedly stopped.
- Digital watch face and standard AAPS watch face have the same issue

Will try the reinstall steps suggested and report back.

maheini, 2023-01-05:
Does your watch have the latest built of AAPS? Have you tried to rebuild & generate the latest phone .apk

I'm using build 2022.12.15 on watch 4 & Huawei phone - everything is working - but just after I installed the latest build onto the watch.

old-square-eyes, 2023-01-05:
I'm using 3.1.0.3

vanelsberg, 2023-01-09:
Will testing dev I have been updating my Watch4 with new .apk versions multiply times without anu problems.
Are you absolutely sure you did not by accident install an new version .APK that was build with a different signing key?
This think this could maybe cause a duplicate installation, hence problems?

old-square-eyes, 2023-01-09:
100% sure on that. I built both the apks at the same time with the same key and stored them in my Google Drive where they are on thier own. Also it worked for a month. It's working again after a reinstall.

wootmasterslick, 2023-01-16:
Same problem here with Galaxy watch 5. Works for short time after reboot of watch then it stops

Aeliassen, 2023-02-21:
I have the same problem running Galaxy Watch 5 Pro. 
When using stock watchface the app stops sending commands (calculator, treatments tt). After restart of the watch the app works fine for a short while. The watch still recieves and displays data from AAPS (BG, IOB, COB) 

I don't think this issue happens if I use a AAPS watchface

agmsd, 2023-03-16:
Yep! Not working for me as well!

hartlijn, 2023-09-12:
Same problem here. Samsung watch 6, WearOs4.
When changing back to AAPS watchface and clear cache of AAPS wear, it works again. But not with another watchface

old-square-eyes, 2023-09-12:
Seems worse when using another watchface too: https://www.facebook.com/groups/AndroidAPSUsers/posts/3653060784915312/

olorinmaia, 2023-09-12:
https://github.com/nightscout/AndroidAPS/issues/2336 seem to be same issue. Have anyone tested this in dev/RC1?

Philoul, 2023-09-15:
Hi,
I confirm this issue on my side:

When AAPS Watchface is selected, everything is fine

When another watchface is selected (I selected one free watchface with complication and also put aaps complication within this external watchface) then:
- immediatly after sselection tiles work
- after a little while (about 30min during my tests), when I select an action from tile, after validation I see the toast message but no validation screen received from aaps.
- when I switch back to aaps watchface, immediatly after, tiles doesn't work but after a few min, then communication restart to work correctly.

This sequence applied twice today reproduced exactly the same result...

Test done with latest dev (aaps and watch side)
Galaxy watch4.

I can help understanding where could be the issue (but I have a full released apk on my watch so with log message and logcat analysis...) some ideas where?
Issue within rxbus sent from watch in tiles activities?

Philoul, 2023-09-15:
When I check aapsLogger into DataLayerListenerServiceWear (on sendMessage)
=> Every minutes, a message is sent on /rx_bridge for ActionHearRate if an AAPS Watchface is selected
=> When I switch to another Watchface, these messages is no more sent immediatly
=> When I switch back to AAPS Watchface, a new HeartRate is sent exactly one min later

See below status of ActionHearRate...
First switch from AAPS to another WF at exactly 01:17:30
Switch back to AAPS WF at 01:19:58
Switch back to other WF at 01:22:45
Switch back to AAPS WF at 01:23:18

```
01:10:44.627  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":59999,"timestamp":1694819444597,"beatsPerMinute":60.057700961682706,"device":"samsung SM-R890"}
01:11:44.619  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":60000,"timestamp":1694819504597,"beatsPerMinute":68.09698469465583,"device":"samsung SM-R890"}
01:12:44.632  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":59999,"timestamp":1694819564596,"beatsPerMinute":58.45579211445232,"device":"samsung SM-R890"}
01:13:44.613  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":60001,"timestamp":1694819624597,"beatsPerMinute":63.5341244312595,"device":"samsung SM-R890"}
01:14:44.628  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":60000,"timestamp":1694819684597,"beatsPerMinute":61.69691165502747,"device":"samsung SM-R890"}
01:15:44.608  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":59999,"timestamp":1694819744596,"beatsPerMinute":68.29785181937744,"device":"samsung SM-R890"}
01:16:44.643  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":60001,"timestamp":1694819804597,"beatsPerMinute":72.97246133328711,"device":"samsung SM-R890"}
01:20:58.480  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":60033,"timestamp":1694820058453,"beatsPerMinute":56.6060195701555,"device":"samsung SM-R890"}
01:21:58.483  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":60000,"timestamp":1694820118453,"beatsPerMinute":58.352866666666664,"device":"samsung SM-R890"}
01:24:18.611  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":60045,"timestamp":1694820258577,"beatsPerMinute":60.137160956638525,"device":"samsung SM-R890"}
01:25:18.607  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":60001,"timestamp":1694820318578,"beatsPerMinute":61.79001914090117,"device":"samsung SM-R890"}
01:26:18.613  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":60005,"timestamp":1694820378583,"beatsPerMinute":64.00511899111804,"device":"samsung SM-R890"}
01:27:18.604  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":59994,"timestamp":1694820438577,"beatsPerMinute":65.66088651580644,"device":"samsung SM-R890"}
01:28:18.599  D  [main]: [DataLayerListenerServiceWear.sendMessage():178]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.ActionHeartRate","duration":59999,"timestamp":1694820498576,"beatsPerMinute":63.46273539332509,"device":"samsung SM-R890"}
```

old-square-eyes, 2023-11-23:
On 3.2.0.2 where I know there may have been improvements. Unfortunately still seeing this for both native and third party watch faces.

M0P, 2023-12-21:
I had the same problem with third party watchfaces. Since it bothered me, I tried some random things I found on the internet and chatGPT suggested (I am not a developer) and currently I (or better ChatGPT) had an idea that works for me so far. I have been using my watch for 6 hours with a third party watchface and everything still works.

I changed this line of code (WearApp.kt line 34):
` startService(Intent(this, DataLayerListenerServiceWear::class.java))`

to 
` startForegroundService(Intent(this, DataLayerListenerServiceWear::class.java))` 

I don't know if this breaks anything else. I haven't noticed any problems yet.

I am running AAPS 3.2.0.3 and have a GW6 with WearOS 4.0.

MilosKozak, 2023-12-21:
try latest dev https://github.com/nightscout/AndroidAPS/commit/8836f90a59e29de9f5a6f47a8e546d156c2f2b67
now it uses foreground service

vanelsberg, 2023-12-21:
Interesting!

olorinmaia, 2023-12-22:
Tested one day now and problem seem to be fixed by this change :) @MilosKozak could this be added for a possible 3.2.0.4 release?

old-square-eyes, 2023-12-22:
Also tested for one day and looking good. I have been testing a 3rd party watch face (yet to test native). No issues so far and it worked after the watch charged all night so would have had every opportunity to sleep the app/treatment cards.

I'm super stoked about this one. Thanks @M0P and @MilosKozak.

vanelsberg, 2023-12-22:
> try latest dev [8836f90](https://github.com/nightscout/AndroidAPS/commit/8836f90a59e29de9f5a6f47a8e546d156c2f2b67) now it uses foreground service

@MilosKozak Unfortunately, on my Samsung Watch4 I now get a continuesly "ongoing notification" that can not be cleared.

So I would say this solves posters problem for non-AAPS watchfaces, but introduces a problem for users when using the AAPS watchface (which was operating just fine before this fix.

(Not sure if this is a bug (not closing notification channel) or a result of the latest "Wear: create notification channel" commit?)

old-square-eyes, 2023-12-22:
@vanelsberg interesting. I just switched to AAPS watchface and tested with a bolus and didn't get that.

vanelsberg, 2023-12-22:
@old-square-eyes Did you build latest commit https://github.com/nightscout/AndroidAPS/commit/041d0f0b7bd43042dbf201f442c1a48433eb22cd
?

old-square-eyes, 2023-12-22:
> @old-square-eyes Did you build latest commit [041d0f0](https://github.com/nightscout/AndroidAPS/commit/041d0f0b7bd43042dbf201f442c1a48433eb22cd)
> 
> _Note: my initial message pointed to the wrong commit..._

Oh my mistake. I only built the one right after (https://github.com/nightscout/AndroidAPS/commit/8836f90a59e29de9f5a6f47a8e546d156c2f2b67)

vanelsberg, 2023-12-22:
@MilosKozak @old-square-eyes 
_Confirming problem was introduced by commit_ [Wear: create notification channel](https://github.com/nightscout/AndroidAPS/commit/041d0f0b7bd43042dbf201f442c1a48433eb22cd)

Tested with previous commit #f3be1bbfe1fe78aee30d791ae5f5e6dadab579b1 (updated both AAPS and Wear app):
That build was ok. Problem is not in this commit.

vanelsberg, 2023-12-27:
> @MilosKozak @old-square-eyes _Confirming problem was introduced by commit_ [Wear: create notification channel](https://github.com/nightscout/AndroidAPS/commit/041d0f0b7bd43042dbf201f442c1a48433eb22cd)
> 

Problem was fixed through https://github.com/nightscout/AndroidAPS/issues/3137.
Nice!

old-square-eyes, 2023-12-27:
Just reporting back after a week of use (https://github.com/nightscout/AndroidAPS/commit/8836f90a59e29de9f5a6f47a8e546d156c2f2b67). Issue is significantly reduced.

I have had my watch run out of battery during the day twice, and a couple of times later in the day, where it doesn't matter so much. I have the 44mm Watch 4 variant with the bigger battery and it's a couple of years old. I have also experienced unresponsive treatments a couple of times requiring a watch restart.

Not sure if subsequent commits have changed things much. But over-all I'd say things are usable and significantly better. I'm loving treatments by watch and am finally able to use a 3rd party WF with xDrip + AAPS complications.

[image]

wootmasterslick, 2023-12-27:
> Just reporting back after a week of use (https://github.com/nightscout/AndroidAPS/commit/8836f90a59e29de9f5a6f47a8e546d156c2f2b67). Issue is significantly reduced.
>
>
>
> I have had my watch run out of [...]

Odd that you still have some unresponsiveness.

Would you suggest that battery usage of the watchface is now significantly increased? What do you think is the % difference?

M0P, 2024-01-09:
I have also been using version 8836f90 since I suggested the change here. I have had practically no unresponsive treatments. I also have no persistent notification.

I did not use the 041d0f0 commit. As @vanelsberg says, it may be that issue #3141 was introduced by the 041d0f0 commit.

Since @old-square-eyes and I have no problems with 8836f90, maybe 041d0f0 is not needed at all? Maybe someone with problems can re-test version 8836f90?

For me, the battery life has improved since I can use a custom watch face. But I think that depends a lot on the watchface.

closed 2024-01-24 by MilosKozak