nightscout/AndroidAPS #2325: Enhancement request Automation - Profile change duration

kristinsmotz, 2023-01-04:
Use case: we have a "School day profile" and "Weekend profile", and would like to switch between them by automation. To avoid resetting autosens we would like to use a duration for this. Per now, for an automation profile switch, the only option is to choose a named profile and make a non-durational profile switch, OR use set a percentage profile switch which allows using a duration. I would think those two different automation dialogs could be combined to resemble the "regular" profile switch dialog, i.e. including the following input fields:
Profile (list of profiles)
Duration
Percentage

Then more options would be possible. 
Constraints/allowed limits for duration and percentage could remain the same as they are now.

MilosKozak, 2023-01-05:
if the profile is different, autosens should be reset

kristinsmotz, 2023-01-06:
Another reason separate from autosens reset is that it would be simpler if we could use a duration for the weekend profile (i.e. at 22.00 on Friday, set weekend profile for 48 hours). If not, we will need two automations, one for Friday to set weekend profile, then one for Sunday night to set weekday profile. If we create a new profile (like I recently did when we started using Lyumjev, to have a fallback profile if we need to go back, and have done several times during honeymoon end/puberty/general changes to insulin requirements), then I will now have to adjust the automation for weekday profile to reflect the new profile, as well as adjusting the automation for weekend profile if that one also is new. 

But also it's about making the system more consistent, in a manual profile change we can select profile, duration and percentage in the same operation, so I don't really see why it's necessary to have two different automation setups, one for profile name and one for profile percentage/duration. 

One of my favorite things in AAPS is the ability to have multiple profiles and creating new profiles to counter changes in insulin requirement so it's easy to switch back if the new profile doesn't work as well as intended, in addition to having ready made "sick profile", weekend profile, hormonal cycle-based profiles etc. Instead of having one profile that we keep doing changes to, with the risk of forgetting what changes was made at what point making it harder to backtrack. 

So it makes sense to me that after testing by doing manual profile switch, we could automate some of the profile switches with all the same options that are available during manual profile switch.

Kenoubi, 2023-08-06:
I agree, would like to see a single "change profile" action that lets you optionally set a profile (if not, stays on whatever profile you were already using), set a percent (default to 100%) and optionally set a duration (if not, the change is permanent).  I.e. feature parity between the dialog to change profiles manually and what's possible through automation.

MilosKozak, 2026-10-11:
In the current dev version, automation can run a Scene. A scene can switch to a chosen profile with a percentage for a set time (up to 3 days), and the profile goes back when the scene ends. Use the "Run scene" action for this.

closed 2026-10-11 by MilosKozak