nightscout/AndroidAPS #2326: EOPatch2

Viewquest, 2023-01-04:
I've been testing the EOPatch2 for a few days. The pump works perfectly. She is flawless. However, it would be better for the users if the alarm settings could be adjusted more individually.
Since the pump gives continuous alarms after 3.5 days, an interruption would be necessary here. The other alarms would also make sense with a single deselection or deselection.
Maybe that can be taken into account. Would be very happy.
I use the DEV-F from 01/02/2023.
[image]

sfriedriszik, 2023-01-04:
Dear developers,
I also use this pump and would be very happy about the function to reset the alarm.

MilosKozak, 2023-01-05:
the pump shoud be replaced after 3 days

sfriedriszik, 2023-01-05:
@MilosKozak 
That is not correct. In Germany, the EoPatch is designed to be worn for 3.5 days. The health insurance also calculates the requirement with 3.5 days. Accordingly, only 9 pumps are delivered per month.
According to Menarini, the pump has a wear time of 3.5 days + 12 hours

Glimmerglykose, 2023-01-29:
> the pump shoud be replaced after 3 days

The pump is supposed to be on for 3,5 days, with an added 12 hours if there is still insulin left in the reservoir. A silencing of the alarm that occours at 3,5 days is needed to keep the patch on for the intended amount of days. We will be wasting 12 hours of battery life if we don't have an option to silence the pump or at least set some individual alarms :)

fjpezuela, 2023-03-18:
Hello,

Next Monday I will start using the EOPatch2 pump. The pump is indeed ready to use for 3.5 days with a grace period of 12 hours afterwards.

I have been using Omnipod Dash and it would be great to be able to silence the alarm after 3.5 days, as it can catch you at a bad time for a change (driving, in a meeting, etc).

This improvement would be a great help.

Thank you very much.

eduardomarron, 2023-05-03:
> Hello,
>
> Next Monday I will start using the EOPatch2 pump. The pump is indeed ready to use for 3.5 days with a grace period of 12 hours afterwards.
>
> I have been using Omnipod Dash and it would be [...]

Totally agree, during the grace period we should be able to silence the alarm or extend the notification to every 2 hours for example. Otherwise the pump will beep quite often and it's quite annoying when you're in class or at a meeting etc.

miguelavh, 2023-11-17:
In official APP (for glucomen day pump) Narsha It's possible to unable this sound, would It be posible to implement this feature in this great work?

closed 2023-11-17 by Viewquest