nightscout/AndroidAPS #2329: AAPS crashes when applying the setting

Viewquest, 2023-01-06:
I have the following error since the last DEV F version: If I go to settings, AAPS is terminated immediately. All other functions are still applicable. In addition, since the penultimate version, the Eop2 has been beeping when I make a profile change. Attach to the last LOG.
[AndroidAPS_LOG_1673011924856.log.zip](https://github.com/nightscout/AndroidAPS/files/10360491/AndroidAPS_LOG_1673011924856.log.zip)

MilosKozak, 2023-01-06:
what is last dev version?

Viewquest, 2023-01-07:
[image]

Viewquest, 2023-01-07:
AAPS only crashes when I press the Settings button.
[image]

MilosKozak, 2023-01-07:
there is already newer commit
[image]

try latest dev

Viewquest, 2023-01-08:
Thank you, I will do it immediately.

Viewquest, 2023-01-08:
Installed the new version and still AAPS crashes when using the "Settings" button.

Viewquest, 2023-01-08:
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/10368170/AndroidAPS.log)

MilosKozak, 2023-01-08:
there is no app start in log

Viewquest, 2023-01-08:
Ok, how can the error be fixed. The version from 01/02/2023 did not have this error.

MilosKozak, 2023-01-08:
do not open all preferences but only plugin preference one by one .
It's probably in some I don't use

Viewquest, 2023-01-08:
Thanks for the hint. I `ll do it. Go to Kodnfiguration and can also set everything for DYNISF. Just think that the error irritates many users.

vanelsberg, 2023-01-09:
Can not repro this (dev-f #4b6eab6afrom, fullrelease, 08/01).
DASH/BYODA, not using DYNISF?

Viewquest, 2023-01-09:
I use the ClucoMen Pump and Xdrip. I no longer use Byoda and Dash.

MilosKozak, 2023-01-09:
open all availavable preferences from Config Builder directly

sfriedriszik, 2023-01-11:
@MilosKozak I have exactly the same problem. From the main overview -> 3 dots menu -> settings AAPS crashes directly.
it only happens with configured EO patch pump. I suspect it's due to the recent changes made to the EO Patch Pump. This problem does not exist in a version from 01/04/2023.

Viewquest, 2023-01-11:
The last two updates now also have the problem with NS. As soon as I want to go to the NS settings, the AAPS crashes. In addition, the data will no longer be transferred to fNS. The error message appears: Error i.o.socket.engineio.client.EngineIOExcception.xhr poll error

Viewquest, 2023-01-11:
[image]

MilosKozak, 2023-01-11:
@Viewquest regarding crash update. you are using old version

Viewquest, 2023-01-11:
Hello Milos, I can't get into NS at all to be able to change something if necessary. Also when I go through the configuration, AAPS crashes.

MilosKozak, 2023-01-11:
I see the crashed on Pixel 4a using old version

Viewquest, 2023-01-11:
[image]
Do you mean the NS version?

MilosKozak, 2023-01-11:
AAPS version

sfriedriszik, 2023-01-11:
@MilosKozak Sorry, it's working again for me at the moment. I'll report back if the error crashes again.

Viewquest, 2023-01-11:
I have the AAPS 3.1.0.3-def-f 11.01.2023

MilosKozak, 2023-01-11:
I see crashes comming from 7ad13882ef 
The data is only date of build, not version

Viewquest, 2023-01-11:
[image]

closed 2023-01-11 by MilosKozak