nightscout/AndroidAPS #2330: Dev F - tdd calculation doesnt work anymore

RalfCGM, 2023-01-08:
Always tdd calculation was slow, but came up eventually. 

Now tdd calculations dont Come up anymore. Only "calculution is being executed". 

Would love to give additional information to provide usefull input. Because of checking profile settings, ive disabled loop almost all the time lately. Could that interfere? 

Ive already pushed reset from the statistics screen. That didnt help. 

Tdd calculation from dana-i screen still work.

MilosKozak, 2023-01-08:
it works well for me. more info needed

RalfCGM, 2023-01-08:
Since im testing a lot with adjust basals, cr and isf,  a lot of profile changes and now about 6 days with running loop disabled is actually the only thing different from when loop was still running. Looks to me that something with the database has gone wrong somehow. Or maybe that loop need to be running to provide the proper data? 

For what it matters. History browser has never worked properly. Really slow or Just hsnging, not providing any data. Dont know whether this could be related to the database being corrupted maybe? 

Which info do you need to be able to investigste? I think the logs might not provide enough data?

RalfCGM, 2023-01-09:
Ive Just ran lgs mode for a while an tdd data was available again. So there must be a relation between the tdd calculation and stopping of some processes when loop is stopped.

Knowing this, this might not be the worst issue. Or does this still indicate an error for you?

MilosKozak, 2023-01-09:
calculating tdd is based only on data in DB. there is no relation to loop

RalfCGM, 2023-01-09:
Then there must be some other connection between running loop and tdd calculation. Im pretty sure that: loop disabled - - > after a couple of hours/days tdd wont calculate anymore. Enable loop - - > tdd calculation works again. Or are there other possible dependecies for tdd not to run?

At first i thought that i found a major issue, but since it looks like enabling loop resolves the issue, most People wont run into this. Because almost nobody Will stop loop as long as i did.

Unless you think there might be some deeper technical issue, we can maybe close this issue to be able to focus more on more relevant issues.

MilosKozak, 2023-01-10:
run it and if it does not appear send recent log

vanelsberg, 2023-01-10:
Was seeing the same in in recent builds of dev-f.
After updating to commit#4b6eab6a (8/1/2023) it started working again.

closed 2023-01-19 by MilosKozak