nightscout/AndroidAPS #2335: AAPS/NSClient watchface improvement - add temp target display option

olorinmaia, 2023-01-10:
Absolutely loving the watch faces in the wear app from NSclient and AAPS. The only thing I'm missing is an option to display temp-target. I know I can see it if I go to settings, but that only applies to AAPS watchface.

That way it will be easier to remember to turn of temp-target if it's displayed on the watch faces when active.

olorinmaia, 2023-10-31:
Don't need to be a visual thing either. Maybe if TT is active text "TT: target-value" could show on custom watch face when TT is active. @Philoul

olorinmaia, 2024-02-18:
Or another option, like on Widget:
If normal target display value with default font-color.
If target is adjusted by autosens or DynIsfSens display value with green font-color.
If temp target display value + (duration in hour and minutes') with yellow font-color.

olorinmaia, 2024-11-20:
@Philoul is this now easier to accomplish with latest changes in dev?

Philoul, 2024-11-20:
It's neither easier nor more complicated...
I just have to take some time to develop it 😉.
=> I have to add 3 views within custom watchface layout (one to manage main user, and 2 additional views for external users now available for the followers), update code to manage these views with text and colors, and update communication to send data.
Maybe add a new parameter to manage if Target should be visible or hidden in watchface (😱 we are trying to reduce them to simplify settings... see topic opened yesterday in ux-design channel 😂)

Then users will have to create new CWF or update existing one to include this new information.

Philoul, 2024-11-23:
See PR #3551 ;-)

olorinmaia, 2024-11-23:
Ty @Philoul

closed 2024-11-23 by olorinmaia