nightscout/AndroidAPS #2336: WearOS entered treatments from Pixel Watch not received by Pixel7 phone (no confirmation)

martiandeath, 2023-01-11:
Using Pixel 7 phone and Pixel watch
AAPS build version 3.1.0.3

APK key is identical and watch shows up in Wear tab on phone
Able to receive info from phone, eg bgl, cob, iob, profile, target, etc
Not using aaps watchfaces
Able to successfully send command from watch to phone once the wear app is started only for a short period of time until something happens and am no longer able to.
When try to send a command the next time, no confirmation message comes up on watch and command is not received by phone.
Must close and restart AAPS on watch. Then able to send commands again for short period of time.
Must close and restart the watch AAPS each time, in order to have confirmation received.

Log file of one-off command sent successfully (added 5g carbs) at 17:16 Aus Eastern Daylight Time (GMT+11):
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/10389433/AndroidAPS.log)

When trying to send command from watch when not working, log file does not receive any updates.

wootmasterslick, 2023-01-16:
Here the same problem with a brand new Galaxy watch 5. Tried multiple watchfaces. 

I find it very odd the function only works for a very short time.

databoy2k, 2023-01-16:
[aapswear.txt](https://github.com/nightscout/AndroidAPS/files/10429492/aapswear.txt)

Same issue. I've attached my ADB logcat which was cleared before I started logging. Everything in this logcat is strictly the following steps:
1) Tap screen to wake up watch
2) Tap AAPS Blood Glucose complication (expected outcome: should go to AAPS menu) - nothing happens on watch
3) Tap AAPS Blood Glucose complication again - still nothing happens on watch
4) Open WearOS apps menu and select "AAPS"
5) Run "Calculator" with 3g carbs, tap checkmark
6) Observe "Calculation Requested" toast
7) Nothing further happens for UI (expected: notification on watch and screen titled "Confirm" with calc. wizard response, in my case it recommended a 0.2U bolus)

This log came from my Galaxy Watch 4 version R890XXU1GVK7 running AWF Infograph watchface, AAPS Wear B724620953-2023.01.13 (dev-g), AAPS phone b724620953-2023.01.13 (fullrelease).

I've noted that I don't have the same issues as long as I'm running any AAPS watchface (e.g. AAPS(DigitalStyle)). Calculator works consistently and I can control the phone.

wootmasterslick, 2023-01-23:
Is anybody looking into this?

mk89pwnz, 2023-04-05:
Can confirm this, had it with GW4 and now with GW5 Pro, only way is to use AAPS watchface, I even tried to force keep running the aaps watch app but still the same problem, maybe wearOS is killing it if watchface is not used? After some time I just gave up and use aaps watchface :)

bedevos, 2023-04-24:
Same problem here.

jchester47, 2023-11-10:
I am having the same issue on my Galaxy Watch 6 Classic when using any watchfaces that are not the default ones available in APS and sent to the phone. If I use a Galaxy watchface or a custom one, the complications work but the commands panel does not. 

If I force close and restart the AAPS app on the watch, the commands work for about 10 minutes but then stop working. 

There seems to be some sort of odd dependency between the watchface and the phone commands. 

This is very annoying since I have a watchface I prefer over the defaults.

mk89pwnz, 2023-11-10:
Same it's and old bug and imo wearos is killing aaps on watch unless it's being used, I would love to use simple pixel like watch face with just compilations form aaps. With 3.2 you can make your own watchface but it will take a lot of time

jaredwhittle, 2023-11-11:
I have this issue too. Even though I open the app on the watch, which should reactivate the app and pull it out of some sort of battery save suspend mode, it still won't connect to the phone app to confirm the treatment.

closed 2024-01-24 by MilosKozak