nightscout/AndroidAPS #2339: AAPS Watch bolus calc improvement - add calc to treatments tab

olorinmaia, 2023-01-11:
AAPS watch bolus calculator is a great feature, especially that you can lock the choices so that caregivers at school wont hit the wrong buttons when bolusing. The only thing missing is the log from the calculation of the bolus when this is done from the watch.

Is this something that could be possible to improve so that bolusing from watch also add "Calc" in treatments tab in AAPS?

Treatment 18:14 is done from AAPS calculator on the phone and 17:15 from calculator on watch

[image]
[image]

MilosKozak, 2023-01-13:
https://github.com/nightscout/AndroidAPS/commit/34ba03bef4ddcd3233edc619af2e812b93889c75

closed 2023-01-13 by MilosKozak

olorinmaia, 2023-01-13:
@MilosKozak I'm pretty new to github and AAPS. Does this mean that the improvement will be implemented in the next master version of AAPS? :)

MilosKozak, 2023-01-14:
yes. now is in dev