nightscout/AndroidAPS #2342: [DEV][NSV3] No connection to NS

jbr7rr, 2023-01-12:
Build: 6fb8339238-2023.01.12 (dev-g) fullrelease
Logs: [AndroidAPS_2023-01-12-15_JB.zip](https://github.com/nightscout/AndroidAPS/files/10402577/AndroidAPS_2023-01-12-15_JB.zip)

Time of occurrence/interest in logs:  14:45

NSClient V3 will not connect to my nightscout

What I have tried already:
- Database reset
- Enabling/disabling ns upload
- different access tokens (created a new one for testing, with admin role)
- Switching back and forth between NSClientV1 and V3

[image]

MilosKozak, 2023-01-12:
14:23:20.538 [DefaultDispatcher-worker-1] E/CORE: [LoadStatusWorker$doWorkAndLog$2.invokeSuspend():27]: Error: 
retrofit2.HttpException: HTTP 404 
	at retrofit2.KotlinExtensions$await$2$2.onResponse(KotlinExtensions.kt:53)
	at retrofit2.OkHttpCall$1.onResponse(OkHttpCall.java:161)
	at okhttp3.internal.connection.RealCall$AsyncCall.run(RealCall.kt:519)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
	at java.lang.Thread.run(Thread.java:1012)

Are you using NS version 14?
PM me config and master password. I'll test

jbr7rr, 2023-01-12:
Yes NS 14.2.6, PM sent

MilosKozak, 2023-01-12:
trailing / in url

closed 2023-01-12 by MilosKozak